openthread/openthread #571: otOpenUdpSocket Requires Caller to use Internal class Ip6::UdpSocket

nibanks, 2016-09-12:
Because the `Ip6::UdpSocket` class now takes a parameter in its constructor (`Udp &aUdp`), code external from the main openthread library cannot call `otOpenUdpSocket` with a plain `otUdpSocket` object. `otOpenUdpSocket` immediately casts any `otUdpSocket` passed in to a `Ip6::UdpSocket` which will result in a null pointer read since the constructor was never called to set `mTransport`.

jwhui, 2016-09-12:
Shall we just have OpenThread maintain a fixed number of sockets for the client (with a compile-time configuration parameter)?

jwhui, 2016-09-16:
In #614, @wbober proposed to initialize the `mTransport` field as part of `otOpenUdpSocket()`, which could work as well.  Thoughts?

nibanks, 2016-09-16:
I'm fine with that approach.

jwhui, 2016-09-20:
Fixed in #623.

closed 2016-09-20 by jwhui