openthread/openthread #574: Suspicious VerifyOrExit() in Local::SendServerDataNotification()

wbober, 2016-09-13:
I'm going through the code and I've some concerns about this condition in `network_data_local.cpp:211`

    VerifyOrExit((mMle.GetDeviceMode() & Mle::ModeTlv::kModeFFD) == 0 ||
                 (mMle.IsRouterRoleEnabled() == false) ||
                 (mMle.GetDeviceState() >= Mle::kDeviceStateRouter) ||
                 (mMle.GetActiveRouterCount() >= mMle.GetRouterUpgradeThreshold()),
                 error = kThreadError_InvalidState);

If any of the conditions evaluates to true the verify is positive (will not exit). So, for example, `mMle.GetDeviceMode() & Mle::ModeTlv::kModeFFD) == 0` doesn't trigger an error. Is that the intended behavior?

In addition:
1) According to Thread 1.1 spec children can act as servers, hence I'm not clear why `mMle.GetDeviceState() >= Mle::kDeviceStateRouter` is needed.
2) What is the reason to check for active router count?

jwhui, 2016-09-13:
Yes, the intent of that logic is to only consider sending a Server Data Notification message if one or more of the listed conditions are true.

> 1) According to Thread 1.1 spec children can act as servers, hence I'm not clear why mMle.GetDeviceState() >= Mle::kDeviceStateRouter is needed.

OpenThread allows children to act as servers.  If the device is not currently a router, then at least one of the other conditions must be true for it to send a Server Data Notification message.

> 2) What is the reason to check for active router count?

If the active router count is above threshold, then this device (if it is a REED) will not attempt to become a router.  If the active router count is below threshold, then we want to delay sending the Server Data Notification message until after becoming a router or enough other routers have become active.

wbober, 2016-09-14:
> Yes, the intent of that logic is to only consider sending a Server Data Notification message if one or more of the listed conditions are true.

Ok.

Personally, I find driving the logic using VerifyOrExit' a bit confusing. I much prefer when its used in sort of assert like checks. But I guess it's a matter of getting used to it :)

> If the active router count is above threshold, then this device (if it is a REED) will not attempt to become a router. If the active router count is below threshold, then we want to delay sending [...]

It's not something I recall from the spec. What's the reason to do this?

jwhui, 2016-09-14:
I understand your concern, the alternative would be to have something like:
```c
if (condition)
{
    ExitNow(action);
}
```
If others prefer this to `VerifyOrExit()` for readability, we can consider moving in that direction.

We suppress sending a Server Data Notification message when a REED is likely to become a Router to avoid frequent updates to the Network Data.  The Network Data is propagated to all nodes in the network (including SEDs if the Server Data is marked stable).  You are correct this behavior is not in the spec and this behavior is currently an implementation-specific optimization.

wbober, 2016-09-15:
The alternative with `ExitNow()` works for me!

Thanks for the clarification.

LuDuda, 2016-09-16:
+1 to use ExitNow() for more complex checks for the readability purpose.

I believe compiler should generate very similar or the same output as one VerifyOrExit macro.

PR #616 (2016-09-16): Change VerifyOrExit to if-statement for readability.
Resolves #574.
closed 2016-09-16 by jwhui