openthread/openthread #576: A server should delay resending ServerDataNotification.

wbober, 2016-09-13:
According to Thread 1.1 spec, a server must wait at least DATA_RESUBMIT_DELAY seconds before resending its data if an inconsistency is detected.

It looks to me that currently this is done upon receiving any advertisement.

jwhui, 2016-09-13:
Thanks for pointing this out.  As a side note, we are currently prioritizing functionality required to achieve Thread Certification.  I don't believe this functionality is covered in the Thread Certification Test Plan.  Regardless, I do believe this functionality should be included.

wbober, 2016-09-14:
Ok I'll have the priority in mind. 

I'm not familiar Thread Certification Test Plan but the Thread spec 1.1 defines this as a MUST. I guess it will get into the test plan at some point but it can wait until then.

jwhui, 2016-09-14:
While Thread Certification is the current priority, we value any feedback of OpenThread's implementation of the Thread Spec.  Not sure if you noticed, but I did submit the PR referenced above to address this specific issue.

wbober, 2016-09-15:
OK.

I'll review the PR.

PR #586 (2016-09-16): Implement DATA_RESUBMIT_DELAY.
Resolves #576.
closed 2016-09-16 by jwhui