openthread/openthread #590: Certification: Output commissioning logs

chshu, 2016-09-14:
Test harness requires a way to get commissioning logs, and retrieval some useful information for verification.

For certification test 8_x_x, harness will verify the following messages based on the logs: 
- JOIN_FIN.req
- JOIN_FIN.rsp
- JOIN_ENT.req
- JOIN_ENT.rsp

Since they are all sent in an encrypted DTLS ApplicationData message.

Logs of the upper four messages need to include following information:
- type (0: JOIN_FIN.req, 1: JOIN_FIN.rsp, 2: JOIN_ENT.req, 3: JOIN_ENT.rsp)
- direction (in/out)
- eui
- length
- tlvs payload

While the **type** is used to verify the present of messages, and **tlvs payload** is used to verify the present of tlvs (only verify Provisioning URL TLV for now, not present in test 8_1_1, should present in test 8_1_6).

chshu, 2016-09-21:
@jwhui, May I ask that do you have any plan of this issue?
I think one way is to store the messages's unencrypted payload, and use a CLI to retrieve them. But  the disadvantage is that it will take some ram space which is very limit now. Please help to advise, thanks.

jwhui, 2016-09-21:
@chshu, I don't have any real plans yet.  However, I'm not sure what is really required.  In looking at #591, do you really need all of that information above?  It seems you only need to log transmitted messages (out direction) and no EUI-64 is required.

chshu, 2016-09-22:
@jwhui, I listed the required information above according to the existing THCI implementation from other vendors, I have attached an example in #591.

But not all of the information are used for verification in Harness test scripts currently, useful information:

- type:  to verify the present of messages
- tlvs payload: to verify the Provisioning URL TLV (only for 8.1.6 Joiner)

So with the workaround in #591, it will help to pass all commissioner tests except 8.1.6 Joiner.

PR #695 (2016-09-27): Enable logging feature for certification test
Resolves #590.

This helps to pass certification test 8.1.6 Joiner, and truly pass 7 commissioning tests which passed based on the workaround.
closed 2016-09-27 by jwhui