openthread/openthread #596: MPL: Updating Seed Set entry

LuDuda, 2016-09-15:
I'm going through the actual codebase of MPL implementation and i have concerns about the check in ip6_mpl.cpp:88.

When MPL Data Message is received with older Sequence Number (according to RFC1982) - MPL module should discard the message, and if I correctly understand RFC7731 (9.3) it should not change any MPL Information Base - especially MinSequence value.

Currently line ip6_mpl.cpp:85 checks only for entry being NULL thus Seed Set will be updated, which i believe is a bug. I can imagine situation where Node A sends MPL Data Messages with Sequence Number 10, 9, 10 so the third packet will be incorrectly processed. This situation i believe is likely to take place, when many routers retransmit some MPL Data Messages. Is it done intentionally? 

If my understanding is correct i may send PR by changing lines 79-82 to be VerifyOrExit macro.

jwhui, 2016-09-15:
Good catch!  Yes, please feel free to submit a PR fix this issue.  Thanks.

LuDuda, 2016-09-15:
Sure I will do PR soon.

jwhui, 2016-09-19:
Fixed in #612.

closed 2016-09-19 by jwhui