openthread/openthread #598: Missing jitters for responses to MLE Parent Request/Link Request

rlubos, 2016-09-15:
Both Thread Specifications and Certification Test Plan define that responses for MLE Parent Requests must be delayed by a random jitter (up to 500 ms for Router only request, and up to 1000 ms for the second request). I'm not sure if this is currenty covered by test plan, it may be in the final version. The same applies for responses to multicast MLE Link Requests (up to 1000 ms).

Also I have a question regarding Mle::BecomeChild function. Why is the initial MLE Parent Request delayed by 1000 ms? Is this a overlook, or are there any implementation-specific reasons for that?

jwhui, 2016-09-15:
> Both Thread Specifications and Certification Test Plan define that responses for MLE Parent Requests must be delayed by a random jitter (up to 500 ms for Router only request, and up to 1000 ms for [...]

This has been a TODO, but low priority because the Certification Test Plan does not test for this behavior.  Are you willing to submit a PR to address this issue?

> Also I have a question regarding Mle::BecomeChild function. Why is the initial MLE Parent Request delayed by 1000 ms? Is this a overlook, or are there any implementation-specific reasons for that?

I don't think there's any reason it should be delayed by a fixed value.  While the Thread Spec does not indicate delaying by a random value, there might be reason to do so (e.g. in the case of merging partitions to avoid a large number of nodes responding to an MLE Advertisement at the same time).  Thoughts?

rlubos, 2016-09-15:
Well, randomizing the initial delay might indeed be beneficial and it seems to me like a good improvement, and as long as it doesn't make tests to fail, it is ok for me.

I will address the missing jitter issue and submit a PR soon.

PR #674 (2016-09-29): Jitter for MLE Parent response and MLE Link Accept
This PR resolves #598.

Here is my proposal for handling delayed responses. This solution introduces DelayResponse class, which is used to store data needed to generate a response and a single timer to manage the pending responses. The maximum number of pending responses is configurable. Currenty Parent Responses and Link Accepts are handled by this mechanism, but it could be easily extended with other messages (e.g. MLE Discover Response should also be delayed according to spec.)

Second commit randomizes the delay before sending initial Pare [... cut]
closed 2016-09-29 by jwhui