openthread/openthread #599: MLE: Seed-id field handling

LuDuda, 2016-09-15:
According to Thread Specification, seed-id field MUST be elided if the IPv6 Source Address is an RLOC, or set to the RLOC16 of the node introducing the MPL data message in the MPL domain otherwise. 

Currently MPL Option has always seed-id present and seed length set to 2 bytes.

Additionally in ip6.cpp:134, even if source address is diffrent than RLOC, last 2 bytes are taken into account, instead of real RLOC16.

For example if i correctly understand, packet send specifically by Application Layer to FF03::1 IPv6 address should have IPv6 source address of ML-EID rather than RLOC, where 2 last bytes of ML-EID be most likely not the same as RLOC16.

jwhui, 2016-09-15:
Yes, I agree this is not spec compliant.  However, in prior testing with other stacks, those other stacks either expected or worked with the 2-byte Seed ID to always be set.  As a result, we have hardcoded the 2-byte Seed ID value for now.

LuDuda, 2016-09-16:
Okay thanks for clarifications. I think that this should be tested by Thread Harness :)

So probably it is valuable to marked this as an issue to be implemented in the future.

jwhui, 2016-09-16:
Thinking about this more, I think we should force the change from Thread 1.0 spec.  Can you submit a PR to fix this?

LuDuda, 2016-09-17:
Yes I can take this task. What I see there are three things to do:
 - TX path: Use seed-id only when IPv6 address is not an RLOC, else add padding (0x01, 0x00) [RFC2460 4.2]
 - RX path: Whenever seed-id is elided, retrieve seed-id from source RLOC address
 - 6LoWPAN: Add support in Extension Header compression/decompression to elide or retrieve padding if needed
 
The last change is needed to make any sense of this change, else we don't save any bytes by eliding seed-id, because of extra padding.

If the angle of attack is acceptable, please give me recommendation about those two issues, since I'm not yet familiar with different layer communication:

 1) How to pass RLOC16 value to the MPL module? Can I pass NetIf pointer, then cast it to ThreadNetIf and use ThreadNetIf->getMle().getRloc16() inside MPL::InitOption?
 I am asking because ideally I would not like to call MLE::getRloc16() function inside generic IP6.cpp or IP6_mpl.cpp files - to not mess with ISO/OSI layer model.
 
 2) Same situation with isRoutingLocator(), unfortunately I need use also Mle::isRoutingLocator(), since Address::isRoutingLocator() checks only IID.
 It may not work if IPv6 global address, generate e.g. by RFC7217 have RLOC form but with different prefix hence we cannot called it RLOC.
 
Also if you are worry about incompatibility with other stacks, I can use some #define for TX path, however I think implementation of RX path should be mandatory. Obviously I don't want to decrease number of passing certification tests.

jwhui, 2016-09-19:
One possibility is to expose an `Mpl::SetSeed()` method that is called by `Mle::SetRloc16()`.  The `Mpl` object could either copy the IPv6 address (which takes 16 bytes) or simply saves a pointer to the `Mle` object's `mLinkLocal16.mAddress`.

I wouldn't worry about the `#define` for now.  We can introduce it later only if absolutely needed.

jwhui, 2016-09-23:
Resolved by #663.

closed 2016-09-23 by jwhui