openthread/openthread #601: MAC: Number of retansmissions

LuDuda, 2016-09-15:
Actually 802.15.4 MAC layer performs up to 15 retransmissions. The value is defined in mac.hpp:72 as kMaxFrameRetries.

Thread Specification mandates to set macMaxFrameRetries PIB (retries allowed after a
transmission failure) to be 3.

Is there any good reason to perform that huge number of retransmissions?

jwhui, 2016-09-15:
The Thread Spec defines `FAILED_CHILD_TRANSMISSIONS` and `FAILED_ROUTER_TRANSMISSIONS` in a way that allows "up to 16 unacknowledged individual frame transmissions and retries".  Currently, OpenThread does not differentiate between MAC and higher layer retransmissions so they are combined as you noticed.  Do you feel that implementing the retransmission logic at both the MAC and higher layer is beneficial?

nibanks, 2016-09-15:
I think it might be worth while. For the Windows implementation, when running OpenThread on host, we are investigating making minor changes to the `Mac` layer to push all backoff and retry logic to the radio layer itself, since Windows cannot guarantee the timer responsiveness that layer requires. If that happens, then it would require the separation of MAC and higher layer retransmission logic, correct?

LuDuda, 2016-09-15:
Thanks for comments! 

How I understand the Thread Specification is that definition of FAILED_CHILD_TRANSMISSIONS and FAILED_ROUTER_TRANSMISSIONS are completely independent from macMaxFrameRetries. Those parameters i would use only when MCPS-DATA.conf for some message returns NO_ACK status, and only for checking if some node has timed-out.

Also I don't see a reason to care about retransmissions of application layer protocol on MAC layer, e.g. MLE specifies optional retransmission pattern (4.9) also CoAP uses its own different one.

jwhui, 2016-09-15:
Okay, I understand the distinction now.  OpenThread's current implementation will transmit a given packet up to 16 times and, if no ack is received, the link will be invalidated.  The Thread spec implies that it should take 4 distinct packets (each with 4 MAC transmissions) before the link is invalidated.  Concretely, if an application is sending packets with a period of X seconds, then OpenThread will invalidate a link on the 1st application packet whereas the Thread specified behavior would invalidate the link after the 4th application packet (3 * X seconds).  Yet another behavior that is not explicitly tested in the cert test plan.  I'm fine with updating `kMaxFrameRetries` to 3 to be inline with the spec.

PR #672 (2016-09-26): Set macMaxFrameRetries to 3.
- Move logic to invalidate Thread links to MeshForwarder.

Resolves #601.
closed 2016-09-26 by jwhui