openthread/openthread #608: Passing sent messages back to upper layers.

jwhui, 2016-09-16:
Currently, when a higher layer component submits a message for transmission, it hands custody off to the lower layers and never gets the message back.  In cases where the higher layer wants to retransmit the same message, it is useful for the lower layer to pass custody back to upper layers.  Both MPL and reliable CoAP transmissions are two examples.

One obvious possibility is for the message header to carry a callback and context that the lower layer could use to pass the message back to the higher layer.  However, this would mean adding two additional pointers to the message header.

Thoughts?

nibanks, 2016-09-16:
Saving the callback & context with the message only works if there are two layers. Is there ever a case where the message needs to be returned through multiple layers?

Either way, I do think it is worth while for a higher layer to get a callback when the send is complete.

jwhui, 2016-09-16:
I was thinking about the multilayer issue as well, but did not identify a concrete need.

rlubos, 2016-09-28:
Hi, I am thinking about extending CoAP implementation on the client side to support retransmissions, non-piggybacked responses etc., and I'm wondering which way to go.

Passing the message back via callback seems to be one solution. In this case there will also be a need to revert the changes made by the lower layers. Maybe it would be needed to store the original size of reserved area in the message header so it could be restored after transmission?

Or, maybe a simple boolean information in the message header from upper layer to the lower one would be enough, whether it should take custody of the message or not. Another bool could inform the upper layer if the message is still being processed in the lower layer or not. But this approach would not allow the upper layer to use list pointers within the message.

Another one, less elegant and more memory consuming is to simply keep a copy of the original message in the CoAP client, and whenever a retransmission is required copy the contents to the another buffer. This way there is no need to pass the message back (CoAP does no seem to require this kind of information). Also the original message buffer could be used to store the retransmission-related data, so there would no need to keep it within the client itself.

What do you think?

jwhui, 2016-09-29:
I agree that maintaining a copy of the message is architecturally cleaner.  I'm okay if the initial implementation takes the copy approach.  Later on, if there is a need to further optimize memory, we can look into eliminating the copies.  I'll close this issue for now.

closed 2016-09-29 by jwhui