openthread/openthread #618: NCP set xpanid & masterkey doesn't work

shahab5, 2016-09-17:
Hi,

I used wpantund to set `Network:XPANID` and `Network:Key` , it reports parse error.

I debugged to find that `spinel_datatype_unpack` uses `SPINEL_DATATYPE_DATA_S` as `pack_format`.
The `ptr` returned points to a string of characters and not integers and as such the length check fails (size of characters is double the expected size).

There should be a conversion from `char` to `int` before proceeding here.

Thanks,
Shahab

darconeous, 2016-09-19:
When you specify `Network:Key`, you need to use the `--data` argument to indicate that the parameter is hexadecimal-encoded binary data:

    wpanctl set Network:Key --data 00112233445566778899aabbccddeeff

shahab5, 2016-09-20:
> Assigned #618 <https://github.com/openthread/openthread/issues/618> to
> @shahab5 <https://github.com/shahab5>.

Thanks for the info.
Will try and update.

Regards
Shahab

shahab5, 2016-09-20:
Verified, its working.
closing this issue.

closed 2016-09-20 by shahab5