openthread/openthread #626: CLI: Routers and children list

LuDuda, 2016-09-19:
I find it a little bit difficult to debug network topology using OpenThread's CLI.

To list all routers or children, two steps procedure is required:

[image]

What do you think about changing behaviour of "router/child list" to something like below? Alternatively to introduce a new one like "router/child table".

[image]
[image]

If yes i will send a PR. The downside is the limit of kTxBufferSize, that is not a problem with current approach.

xiaom-GitHub, 2016-09-19:
@LuDuda The new router/child list looks very pretty. However, current THCI implementation parses and abstracts the key network information from previous output format. Maybe introducing a new cli is better, but not sure if it's a little redundancy for memory resources.

wbober, 2016-09-19:
@xiaom-GitHub 
Maybe a parameter to router/child list command would be an option? For example:
`router pretty-list` or simply `router table`

xiaom-GitHub, 2016-09-19:
@wbober Yes, definitely, that would be a feasible option with a different argument.

jwhui, 2016-09-19:
The list format looks great - feel free to submit a PR that supports the new format with a command-line option.

jwhui, 2016-09-22:
Addressed in #662.

closed 2016-09-22 by jwhui