openthread/openthread #631: Lots of spurious build failures for Cert_5_3_02_RealmLocal.py

darconeous, 2016-09-19:
I've noticed two spurious build failures today, both for `thread-cert/Cert_5_3_02_RealmLocal`.

```
======================================================================
ERROR: test (__main__.Cert_5_3_2_RealmLocal)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/travis/build/openthread/openthread/examples/../tests/scripts/thread-cert/Cert_5_3_02_RealmLocal.py", line 72, in tearDown
    node.stop()
  File "/home/travis/build/openthread/openthread/tests/scripts/thread-cert/node.py", line 138, in stop
    self.pexpect.expect('Done')
  File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 321, in expect
    timeout, searchwindowsize, async)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 345, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 107, in expect_loop
    return self.timeout(e)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 70, in timeout
    raise TIMEOUT(msg)
TIMEOUT: <pexpect.pty_spawn.spawn object at 0x2ab0d8f1f6d0>
command: /opt/python/2.7.10/bin/python
args: ['/opt/python/2.7.10/bin/python', '/home/travis/build/openthread/openthread/tests/scripts/thread-cert/../../../tools/spinel-cli/spinel-cli.py', '-p', '../../examples/apps/ncp/ot-ncp', '-n', '2']
buffer (last 100 chars): "Assertion `GetOffset() + aDelta <= GetLength()' failed.\r\nthread stop\r\nAborted (core dumped)\r\nError\r\n"
before (last 100 chars): "Assertion `GetOffset() + aDelta <= GetLength()' failed.\r\nthread stop\r\nAborted (core dumped)\r\nError\r\n"
```

Full travis build failure log can be found [here](https://gist.github.com/darconeous/6ca6db32ecf83bcac468dab4ac22939e).

jwhui, 2016-09-19:
I'm hoping #628 resolves this issue, but I don't know for sure.

jwhui, 2016-09-22:
Closing because I haven't seen this in a while, we can reopen if we see it again.

closed 2016-09-22 by jwhui