openthread/openthread #639: MLE: Link-Local RLOC" address

LuDuda, 2016-09-20:
I see that mle.cpp defines mLinkLocal16 address, that is registered to the interface when device is a SED (mle.cpp:519). 

What is a use of such link-local address? Thread device configures link-local address only based on its Extended MAC address.

Thread specification explicitly says: "Note that Thread devices do not configure a link-local address based on their 16-bit MAC address."

jwhui, 2016-09-21:
I agree that this is in conflict with the Thread spec.   I believe this is a spec issue that we need to raise with the Thread Group.

The specific scenario we are trying to handle by assigning a link-local RLOC is when a SED is sending IEEE 802.15.4 Data Requests to a parent that has no knowledge of the SED.  The only feedback a SED receives when sending Data Requests is the IEEE 802.15.4 Acknowledgement and a parent will happily acknowledge Data Requests even if it has no knowledge of the SED.  As a result, the parent must be able to send a Child Update Request to notify the SED to reattach.  Because the parent has no knowledge of the SED, the parent can only used information contained in the Data Request itself (e.g. a Source Address which may be a Short Address).

LuDuda, 2016-09-22:
Nice catch! I think definately we need to signal this inside the Thread Group.

I was thought a little about possible solution, and its hard to find any good.

The solution that you may have in OpenThread looks as it have a weak spot in case parent assigns the same Child ID for subsequent children.

Let's say SED1 gets Child ID equal to 1, then parent drops the child and some other
ED attaches. Parent then may assign also Child ID equal to 1 to the new child, causing
potential danger. For example if the new child is not a SED or it is but pulls first and gets some Child Update Response or some other "rejected" message.

Some potential solution would be to not ACKed for 802.15.4 Data Requests for not our's SEDs. But there are two things probably rejecting the idea:
 - All 802.15.4 Data Requests have to be secured (then MIC will failed for old SED)
 - Send ACK to valid children before ACK Timeout - which may look impossible to met since we need to do decryption here.

Or SED may use extended source address when polling.

jwhui, 2016-11-01:
We discussed this topic a few times within the Thread Group.  The group has decided to remove any mechanism for the Parent to notify a given device that it is not a Child.  At the same time, I believe this would also remove any need for a link-local RLOC address.  I'll submit a PR that makes these changes.

PR #913 (2016-11-03): Remove Link Reject and link-local RLOC.
Resolves #639.
closed 2016-11-03 by jwhui