openthread/openthread #651: SLAAC implemented in application.

hubertmis, 2016-09-21:
It seems that SLAAC is implemented in cli application instead of network stack. Is it target design or should be refactored?
Additionally it creates IID randomly while procedure from RFC 7217 should be used.

jwhui, 2016-09-21:
OpenThread intentionally delegates full control of managing IPv6 addresses for prefixes obtained via the Network Data to the application.  We follows this model for a few reasons:

1. We are targeting devices with limited resources.  Ultimately, the application dictates the IPv6 addressing requirements (i.e. what IPv6 prefixes are of interest and how those IPv6 addresses should be configured).
2. While RFC 7217 should be used, there are cases where other IPv6 address formation schemes may be used.  Thread explicitly allows manual configuration of IPv6 addresses as well.
3. In scenarios where OpenThread is operating in an NCP configuration, this model allows the host-side responsible for managing the host's IPv6 interface (e.g. `wpantund`) to maintain control of the IPv6 addresses assigned to the interface.

hubertmis, 2016-09-22:
Maybe we could provide utility in openThread API to help application developer use SLAAC and RFC7217? It would be removed by linker if application did not use it.

jwhui, 2016-09-22:
Sounds good to me.  Are you willing to submit a PR for this feature?

hubertmis, 2016-09-23:
Yeah, I'll prepare PR.

PR #698 (2016-09-29): Pr/slaac utility
Fixes #651
closed 2016-09-29 by jwhui