openthread/openthread #684: Appveyor account & badge

beriberikix, 2016-09-25:
Do we have an OpenThread account on Appveyor? I found @nibanks but none for the OT org. When we're ready, we should create one and add a status page to the top-level readme.

jwhui, 2016-09-25:
We already have one - that's how we get the build status reports for each PR.  I delayed putting it on the README to give appveyor some soak time...

beriberikix, 2016-09-25:
Duh, I didn't think to look at the PR checks. We can keep this as a tracking bug until appveyor is soaked and the badge is up _or_ close since you're all over it.

beriberikix, 2016-09-25:
And to save a search, here's the [instructions for Github](https://www.appveyor.com/docs/status-badges/#badges-for-projects-with-public-repositories-on-github-and-bitbucket).

nibanks, 2016-09-25:
I had already added it to one of my branches to test it out:
https://github.com/nibanks/openthread/tree/windows

jwhui, 2016-09-28:
@nibanks, feel free to submit a PR that adds the appveyor badge.

jwhui, 2016-10-04:
Resolved by #741.

closed 2016-10-04 by jwhui