openthread/openthread #697: Stabilize Cert tests behavior

rlubos, 2016-09-27:
When I was working with the OpenThread code recently, I've encountered a problem with failing thread-cert tests, so I did an experiment with seed generation in posix RNG:
```
void posixRandomInit(void)
{
    time_t now;
    time(&now);
    s_state = (uint32_t)now + NODE_ID;
}
```
After applying this change, some of the tests started to fail at random. Here's an example result:
https://travis-ci.org/NordicSemiconductor/openthread/builds/163050301

I think it might be worth to pull this change and fix failing tests, to minimize the possibility that future PRs will face problems with tests themselves. I can attempt to fix the test and submit the PR if you want.

jwhui, 2016-09-27:
On first glance, it seems many of the failures are due to #554 and increasing the delay when waiting for a device to become a router should solve many of them.  Would be great if you could dig down and make the `thread-cert` tests more robust.  Thanks.

nibanks, 2016-09-27:
I also just finished integrating the latest changes into my Windows driver and now most all the tests seem to fail for me. Generally with something like the following:
```
Traceback (most recent call last):
  File "scripts\Cert_5_1_01_RouterAttach.py", line 67, in test
    self.assertEqual(self.nodes[ROUTER].get_state(), 'router')
AssertionError: 'child' != 'router'
```

I am working on getting the logs for a single repro.

jwhui, 2016-09-27:
@nibanks @rlubos , I'll submit a PR that increases the timeouts just to get issues related to #554 out of the way.  For any remaining failures, would be great to get to the bottom of them.

nibanks, 2016-09-27:
Here are all the logs I have:

[console_output.txt](https://github.com/openthread/openthread/files/496095/console_output.txt)
[logs_filtered.txt](https://github.com/openthread/openthread/files/496096/logs_filtered.txt)
[logs.txt](https://github.com/openthread/openthread/files/496097/logs.txt)

jwhui, 2016-09-27:
@nibanks, looking at your logs, it seems that you need to update your test scripts to configure the router selection jitter to 1.  See #554 for the changes.

nibanks, 2016-09-27:
I have, I just didn't add a log for that yet. In the filtered logs, you should see something like:

`[otNodeApi]     ---> otNodeSetRouterSelectionJitter`

jwhui, 2016-09-27:
@nibanks, okay - I was looking at `console_output.txt` and didn't see the command there.

rlubos, 2016-09-27:
@jwhui @nibanks Just as you say, the most common problem seems to be too short delay when waiting for router promotion after introducing router selection jitter. I applied some fixes in my PR #674, as I was facing this problem, but without randomizing the RNG seed, each test run ends up with the same values returned by the RNG, so I might've not cover every failing test. I'll take closer look into it tommorow and try to fix other tests.

nibanks, 2016-09-27:
Ok, so I figured out there was a bug related to the router selection jitter. It was getting set in the test, but not actually applied in the driver. With that fixed and #700 I think we should be good for now.

xiaom-GitHub, 2016-09-28:
@nibanks Could you please let me know the bug you figured out about router selection jitter? Thanks.

nibanks, 2016-09-28:
The bug was in my own code, not OpenThread.

rlubos, 2016-09-28:
I did a few test runs and they seem to behave quite stable now. The only failure I've encountered was Cert_5_5_5_SplitMergeREED, where one of the routers was not reachable (according to it's RLOC it either did not promote to a Router and for some reason is not accesible, or it did not update it's RLOC after promotion and therefore is not accesible). Unfortunately i was not able to reproduce the issue locally. Either way the problem does not seem to refer to the test code itself. I attached Travis log from the failed run.

[failed_Cert_5_5_5_SplitMergeREED.txt]
(https://github.com/openthread/openthread/files/497736/failed_Cert_5_5_5_SplitMergeREED.txt)

Do you want me to PR the posix RNG update, or shall we leave it as it is?

jwhui, 2016-09-29:
I think it might make sense to seed the RNG with a random value.  Doing so will make the results more non-deterministic.  If it becomes difficult to debug, we may consider logging the seed or making that an optional command-line argument.  Also, rather than using time, shall we just use the `rand()` function instead?

rlubos, 2016-09-30:
Yes, I realized that `NODE_ID` has to be multiplied to avoid overlaps and did it that way on my test branch. I'll submit a PR then.

rlubos, 2016-12-27:
This topic seems to be outdated, I think it can be closed.

closed 2016-12-27 by rlubos