openthread/openthread #711: Create example and document windows suppoer

beriberikix, 2016-09-28:
When we're ready we should create an example demonstrating Windows support. `examples/platforms` seems like a natural place to put the example with either an inline README or Wiki page for documentation. I'd personally like to see it running on Windows IoT Core and an RPi3 if that's technically possible.

@nibanks is that something you're planning on doing?

nibanks, 2016-09-28:
I would definitely like to see that demo as well! Right now I have a lot of Windows related code still in my own branches, currently blocked on a number of PR I have outstanding (and a few queued up waiting for those as well).

The code I have creates a Windows driver that will run OpenThread in Windows and talk to a HW device's Radio interface. I have it working a virtual device, but haven't hooked it up to actual HW yet.

I also plan on supporting the model of OpenThread running on the device as well, using the Spinel interface to talk to it, but that work has yet to be completed.

Bottom line, we are still a little ways to go before we can have a real demo set up.

nibanks, 2016-11-29:
With custom Windows Iot Core images (for the FTDI drivers) I have successfully got everything working with OpenThread running on the CC2538 device, connected to both a Raspberry Pi 2 (ARM) and MinnowBoard MAX (x64).

I'm working out the process to get my changes prebuilt into the IoT Core images, but it might take several weeks or months for it to be in the public images. Once I get all my PRs merged, I will update the documentation.

jwhui, 2016-11-29:
Great to hear, awesome work, @nibanks!

beriberikix, 2016-11-29:
Awesome! Out of curiosity, are there any usb dongles that you know of that are supported by Windows Iot core with the in the in-box [Usbser.sys](https://msdn.microsoft.com/en-us/library/windows/hardware/dn707976(v=vs.85).aspx)? Cursory research suggests that one _could_ hook up [MCP2200](http://www.microchip.com/DevelopmentTools/ProductDetails.aspx?PartNO=adm00393) to the serial to circumvent the FTDI.

nibanks, 2016-11-29:
The Windows IoT Core image already has the FTDI drivers on it, but I just had to update the INF files to add Texas Instrument's PIDs to it. I know there are a number of other supported devices already. I found this [Serial UART](https://developer.microsoft.com/en-us/windows/iot/Samples/SerialUART.htm) sample that recommends [this device](https://www.amazon.com/gp/product/B00LODGRV8) (down in the 'Using USB-to-TTL Adapter' section) for the sample.

nibanks, 2016-12-01:
I have added the initial Wiki page for [OpenThread on Windows](https://github.com/openthread/openthread/wiki/OpenThread-on%C2%A0Windows) and just went through the steps on my MinnowBoard MAX connected to a TI CC2538. With these instructions, anyone with a CC2538 already setup and installed should be able to get the device working on any Windows 10 PC.

Though all the code isn't merged yet, I think we can probably call this Issue resolved.

closed 2016-12-06 by jwhui