openthread/openthread #713: Implement MPL delayed retransmissions.

jwhui, 2016-09-28:
The current MPL implementation retransmits MPL messages only once and without delay.  The implementation needs to be enhanced such that a router retransmits each MPL message multiple times with a randomized delay.

LuDuda, 2016-09-29:
I am willing to take this task. And if i have a green light i can start to work on it from the next week.

jwhui, 2016-09-29:
@LuDuda, thanks for picking up this task.

LuDuda, 2016-09-30:
Just one question. As we agreed here #608 in the first iteration we will do a copy in the application layer (CoAP/MPL) for any retransmission mechanisms. Should we use IP message pool (Ip6::mMessagePool) for this purpose or create a separate configurable message pool in both CoAP and MPL. Personally i believe that eventually in the future we should optimize it in the way that message will not be freed by lower layers when its somehow marked by application, so, for simplicity using IP message pool looks like a good idea for now.

jwhui, 2016-09-30:
@LuDuda, I agree with your thoughts.  Using the IP message pool seems like the right long-term direction.

LuDuda, 2016-10-28:
I think this can be closed.

jwhui, 2016-10-28:
Resolved by #827, #887, and #893.

closed 2016-10-28 by jwhui