openthread/openthread #728: Endless Loop of Tasklets

nibanks, 2016-09-29:
I have the following code:
```
do {
    // Process all tasklets that were indicated to us from OpenThread
    otProcessNextTasklet(pFilter->otCtx);
} while (otAreTaskletsPending(pFilter->otCtx));
```

I have occasionally found that this gets stuck in an endless loop of processing tasklets. I haven't been able to grab it in the debugger yet, and I want to add a count and assert in DEBUG mode to assert if I hit some number of continuous executions of this loop.

1. First, is this code correct? Should I be using this differently?
2. What would be a good threshold to use to treat as a bug if I get more than that many tasklets to execute in a row?

jwhui, 2016-09-29:
This code could be problematic if OpenThread is using the task queue to "busy-wait" until some condition variable changes.  The Mac class had an example of this, but was recently removed in #706.  In general, you should make sure to process any interrupts from platform drivers - doing so will ensure that you don't get stuck when OpenThread is waiting for a state change/event in the platform drivers.

Of course, if OpenThread is not waiting on a state change from the driver, then you may be running into an actual bug that involves endless task processing.

nibanks, 2016-09-29:
My most recent hit included #706, so I am leaning towards a bug somewhere. I have a thread completely dedicated to running and processing OpenThread events (see [otLwfEventWorkerThread](https://github.com/nibanks/openthread/blob/win/driver/examples/drivers/windows/otLwf/eventprocessing.c#L635)). It processes all the events for running OpenThread. How am I supposed to prioritize between them?

nibanks, 2016-09-29:
The issue seems to be related to energy scans:

```
1: kd> ?? pFilter->otCtx->mIp6
class Thread::Ip6::Ip6
   +0x000 mRoutes          : Thread::Ip6::Routes
   +0x010 mIcmp            : Thread::Ip6::Icmp
   +0x040 mUdp             : Thread::Ip6::Udp
   +0x058 mMpl             : Thread::Ip6::Mpl
   +0x110 mMessagePool     : Thread::MessagePool
   +0x40130 mTaskletScheduler : Thread::TaskletScheduler
   +0x40140 mTimerScheduler  : Thread::TimerScheduler
   +0x40148 mForwardingEnabled : 
   +0x40150 mSendQueue       : Thread::MessageQueue
   +0x40160 mSendQueueTask   : Thread::Tasklet
   +0x40180 mReceiveIp6DatagramCallback : 0xfffff805`53ae3df0     void  otLwf!otLwfReceiveIp6DatagramCallback+
   +0x40188 mReceiveIp6DatagramCallbackContext : 0xffffc18c`a559a950 Void
   +0x40190 mIsReceiveIp6FilterEnabled : 
   +0x40198 mNetifListHead   : 0xffffb08a`0b3cc028 Thread::Ip6::Netif
   +0x401a0 mNextInterfaceId : 2 ''

1: kd> dx -r1 (*((otLwf!Thread::TaskletScheduler *)0xffffb08a0b3cbfb0))
(*((otLwf!Thread::TaskletScheduler *)0xffffb08a0b3cbfb0))                 [Type: Thread::TaskletScheduler]
    [+0x000] mHead            : 0xffffb08a0b3ccbe0 [Type: Thread::Tasklet *]
    [+0x008] mTail            : 0xffffb08a0b3ccbe0 [Type: Thread::Tasklet *]

1: kd> dx -r1 (*((otLwf!Thread::Tasklet *)0xffffb08a0b3ccbe0))
(*((otLwf!Thread::Tasklet *)0xffffb08a0b3ccbe0))                 [Type: Thread::Tasklet]
    [+0x000] mScheduler       : 0xffffb08a0b3cbfb0 [Type: Thread::TaskletScheduler &]
    [+0x008] mHandler         : 0xfffff80553afa510 [Type: void (__cdecl*)(void *)]
    [+0x010] mContext         : 0xffffb08a0b3ccad8 [Type: void *]
    [+0x018] mNext            : 0x0 [Type: Thread::Tasklet *]

1: kd> u fffff80553afa510
otLwf!Thread::Mac::Mac::HandleEnergyScanSampleRssi [c:\users\nibanks\source\repos\openthread-win\src\core\mac\mac.cpp @ 230]:
fffff805`53afa510 48894c2408      mov     qword ptr [rsp+8],rcx
fffff805`53afa515 4883ec28        sub     rsp,28h
fffff805`53afa519 488b4c2430      mov     rcx,qword ptr [rsp+30h]
fffff805`53afa51e e84dffffff      call    otLwf!Thread::Mac::Mac::HandleEnergyScanSampleRssi (fffff805`53afa470)
```

jwhui, 2016-09-29:
Yes, `Mac::HandleEnergyScanSampleRssi` is another example of "busy-looping" in a tasklet waiting for an event from the platform drivers.  This implementation is designed to sample the radio's RSSI as quickly as possible and stop when `Mac::HandleMacTimer()` is called.  Of course, the timer handler is only called when the platform alarm fires.  The current OpenThread examples always processes the platform drivers before each call to `otProcessNextTasklet()`.

nibanks, 2016-09-29:
Ok. I put some code to only process up to 10 tasklets in a row before breaking out and trying doing some other things (timers, send/recv, ...). But can you explain a bit more why OpenThread needs to sit in a busy loop, polling? Is this something we might be able to offload to the radio with some kind of get and async completion callback?

abtink, 2016-09-29:
@nibanks, this is a common pattern to have a task post itself back while waiting for some end condition. Note that this is not a pure busy loop, as other tasks may be posted and executed in between. This  pattern is used in `energy-scan` implementation where we want to continuously (and as quickly as possible) sample RSSI for certain interval. It is also used (sort of indirectly) in `NcpUart` class for sending data over uart. 

As @jwhui  suggested, your `do-while` loop should include the processing of drivers (timers, send/recv, etc.) in the same loop where the tasks are being handled. Or they should happen fully from a different thread/context.

nibanks, 2016-09-29:
I totally understand for OpenThread running on the device to using a polling mechanism. But when it is running on a host, such as Windows, not only is it detrimental to the rest of the OS to be spinning, but since the actually `otPlatRadioGetRssi` call will take a while to get to the device and back, it's not going to give the best results.

At least when running on Windows, we would need to offload this entire process to the radio layer. If we don't want to offload this when running on the device, we could use `otPlatRadioGetCaps` to control the OpenThread logic to decide whether to offload the logic or do it locally.

darconeous, 2016-09-30:
> I put some code to only process up to 10 tasklets in a row before breaking out and trying doing some other things (timers, send/recv, ...). But can you explain a bit more why OpenThread needs to sit in a busy loop, polling? 

To the extent that something is using tasklets to implement polling for a condition (which, with your loop, results in a busy-wait), I believe that is a bug that should be fixed. However, I don't think that is the bug we are seeing here.

I think the real problem is in how we dispatch tasklets. Ideally, at each pass through the main loop we would dispatch all tasklets that have been added *up to that point*. Any new tasklets added as a part of dispatching those tasklets would be handled at the next iteration through the main loop. This prevents the other parts of the main loop from being starved of cycles. Unfortunately, I don't think the current API allows you to implement that sort of behavior.

Perhaps we need to adjust the API to allow that?

darconeous, 2016-09-30:
Maybe something like `otProcessTasklets()`. It would only dispatch tasklets queued up at the time it is called. Any new tasklets queued up while the tasklets are being dispatched would be handled at the next main loop iteration.

abtink, 2016-09-30:
The idea of having `otPlatRadioGetCaps` to check if the radio provides an energy scan function sounds good to me. But we will still need to support this for radio platforms that don't provide this feature.

nibanks, 2016-09-30:
@darconeous, I like the idea of changing the API (or just adding an additional one?) to process all currently queued tasklets. When I originally put together my code, I assumed that was what the API was going to do anyways. It took me a while to debug why everything wasn't working...

@abtink, yeah, supporting both sounds good.

jwhui, 2016-09-30:
How about `otProcessQueuedTasklets()`?  Do we want to keep the existing `otProcessNextTasklet()`?

nibanks, 2016-09-30:
I'd vote to not keep `otProcessNextTasklet` since having some clients execute tasklets differently will likely just make things harder to debug if a problem arises in this area. Also, I like the name `otProcessQueuedTasklets`.

PR #735 (2016-09-30): Process all currently queued tasklets in one shot.
Resolves #728.
closed 2016-09-30 by jwhui