openthread/openthread #731: Implement CoAP client

rlubos, 2016-09-30:
If there is noone already working on this tpoic, I'm willing to implement CoAP client with support for retransmission etc.

One question though - shall we provide an API for application to create/send CoAP messages, similar way it is done with a raw UDP?

jwhui, 2016-09-30:
A proper CoAP client would be a great enhancement.  I think it makes sense to create an API that allows applications to use OpenThread's CoAP implementation.

darconeous, 2016-09-30:
We could always use a trimmed-down version of [SMCP](http://darconeous/smcp).

rlubos, 2016-10-03:
I'll take a look at SMCP, thanks.

rlubos, 2016-10-20:
I've submitted PR with refactoring to support CoAP Client (#851). Although there are two troublesome cases which I wanted to discuss.

1. Joiner Entrust CoAP message - according to Thread Specification v 1.1 it shall be secured on MAC with Key ID Mode 0, using single-use KEK. But it still is a confirmable message, and CoAP may need to retransmit it. So the question is - shall we use this key to retransmissions as well, and invalide it on response/timeout, or shall we  ignore CoAP mechanisms in this case and transmit the message only once?

2. CoAP messages transmitted over secured DTLS connection. Current implementation of CoAP client does not support it, so I did not refactor classes that use this feature, but I think it should be handled by the client as well. So the idea is:

    a) Extend current implementation of CoAP client to support establishment of DTLS connection and secure transactions.

    b) Create CoapSecureClient class inherited after CoapClient, and implement new features there.

The former solution might save some resources (no extra timer or queue) but it would overcomplicate the logic IMHO. The latter solution might use some more resources, but separates the logic nicely. Personally I would prefer it this way. What do you think?

jwhui, 2016-10-20:
@rlubos, I also prefer approach 2b.  Consolidating clients behind a single UDP socket structure already saved a chunk of resources, I think it's fine to reclaim a small chunk of that 😄 .

rlubos, 2016-10-21:
I'll implement secure CoAP client then. Shall I provide secure CoAP server as well (I'm willing to if there's positive response)? We could also provide a CoAP server API to provide an application full CoAP experience.

closed 2016-11-16 by rlubos