openthread/openthread #760: Package Managers

beriberikix, 2016-10-06:
I'd like us to consider adopting a package manager to make it easier for users to install/update OT. We definitely can look into after #37 but I think it is generally a good thing.

There isn't a defacto package manager for embedded device software but the closest x-platform one is [PlatformIO](http://platformio.org). I was waiting until 3.0 was released since now I think it will work well for OT. I like it because it doesn't force any [requirements](http://docs.platformio.org/en/stable/librarymanager/creating.html#creating-library) on code structure and we can self-publish libraries, even one-off libs we need in `third_party`. Another useful feature in 3.0 is the ability to define libraries via [`downloadURL`](http://docs.platformio.org/en/stable/librarymanager/config.html#libjson-downloadurl), so silicon vendors _could_ share their BSPs as dependencies directly from their own hosted sites. There's also a set of premium features that might be useful but this free-to-use service could be a great addition. I also had a good experience getting support from the team when I was working on an open source project in the past.

beriberikix, 2016-10-07:
FYI - I started a [thread](https://community.platformio.org/t/support-802-15-4-embedded-platforms/861) on the PlatformIO Community.

beriberikix, 2017-06-04:
Conan was also suggested in #1873

jhwoodyatt, 2018-01-19:
I'm withdrawing my suggestion of Conan in #1873. I've never heard of PlatformIO, and I will look into it. I've been experimenting with another one lately, which I like very much, but I'm not ready to promote it here.

jhwoodyatt, 2018-02-01:
I wanted to test the claim that PlatformIO is open source by reviewing the license. I found the subscription license for PlatformIO Plus, but it was not easy for me to tease out what features of the open sourced logic are independent of the subscription service. If there were alternative hosting services for PlatformIO, and open source development of the PlatformIO Core were not dominated entirely by the PlatformIO Plus fee for service enterprise, this would be less of a concern. Maybe someone more familiar with how PlatformIO is organized can respond to my concern.

jwhui, 2019-02-19:
Closing stale issue. We can re-open if there is renewed interest.

closed 2019-02-19 by jwhui