openthread/openthread #765: Pending Scan Logic in Mac Class Interferes with Any Active Scan

nibanks, 2016-10-07:
In [`Mac::Scan`](https://github.com/openthread/openthread/blob/master/src/core/mac/mac.cpp#L171) if the state isn't currently `kStateIdle` then it ends up setting the `mPendingScanRequest` variable which will be used after the current task is complete.

It also sets all other `mScan*` variables which could currently be in use by a different scan. I believe all those variables need to be duplicated to save for the pending scan, and then copied to the active scan variables when it is no longer pending.

Additionally, if either of the scans are started twice while a not `kStateIdle` the second scan will overwrite the pending variable state.

jwhui, 2016-10-07:
In my reading of `Mac::Scan`, the following checks are performed before making any side effect:

1. `mState` is neither `kStateActiveScan` nor `kStateEnergyScan` (i.e. a scan operation is not currently in progress.
2. `mPendingScanRequest` is `kScanTypeNone` (i.e. a scan operation is not currently pending).

Can you detail the code path that would lead to conflicts?

nibanks, 2016-10-07:
Ah, you are right. I looked for a state check in openthread.cpp, but somehow glanced over the first lines of Mac::Scan. Sorry for the false alarm.

closed 2016-10-07 by nibanks