openthread/openthread #779: Certification: Missing Active TimeStamp TLV in data response packet

chshu, 2016-10-09:
Certification test 8.3.1(Leader) fails since there is no Active TimeStamp TLV in data response packet.

[Leader_8_3_1.zip](https://github.com/openthread/openthread/files/518265/Leader_8_3_1.zip)

chshu, 2016-10-09:
The root cause of this issue is that Leader does not have valid `Active Operational Dataset` by default, so it will not contain Active TimeStamp TLV in data response packet.

In current OT implementation, Leader will get `Active Operational Dataset` either from CLI configuration or from MGMT_ACTIVE_SET.req message, but they are not triggered in test 8.3.1.

To resolve the issue, I think we have two options:

1. Configure a default `Active Operational Dataset` for each Leader;
2. Include Active TimeStamp TLV with value 0 in data response packet if there is no valid `Active Operational Dataset`.

@jwhui @luwannest @xiaom-GitHub , Please help to give your suggestion, thanks.

xiaom-GitHub, 2016-10-10:
@chshu Including Active Timestamp TLV with 0 may be better. Considering network data TLV, it's similar, also empty if no CLI triggers or server device sends server data notification to Leader. In DataResponse message, we can see that Network data TLV is included with 0.

jwhui, 2016-10-10:
I agree, I think responding with an Active Timestamp of 0 is the simplest and best path forward.

chshu, 2016-10-10:
Ok, thanks. I will push PR soon.

PR #782 (2016-10-11): Set active timestamp to 0 if there is no valid active operational dataset
Resolves #779.

With the help of #743 , we have passed certification test 8.3.1(Leader):
[Leader_8_3_1.zip](https://github.com/openthread/openthread/files/519253/Leader_8_3_1.zip)
closed 2016-10-11 by jwhui