openthread/openthread #801: Commissioning Dataset MUST be included while sending stable Network Data to SED

lluwang, 2016-10-13:
This is to pass 9.2.6 (DUT as Router).

jwhui, 2016-10-13:
Is this to satisfy Step 15 of 9.2.6?  I actually think the specs are wrong in this case.  The SED is configured to only receive stable network data.  Step 15 indicates that the Commissioning Data TLV is not marked as stable.  That information is not useful to the SED and should not be provided.

jwhui, 2016-10-13:
FYI, I found [TESTPLAN-272](https://threadgroup.atlassian.net/browse/TESTPLAN-272) which already discusses this issue.

lluwang, 2016-10-13:
Yes. And also file JIRA DEV-1016 for this.
But I also found that 8.4.3.2 said that "When transmitting stable thread network data, Thread Network Data MUST include a Commissioning Data TLV with the Stable flag set to one."
How do I understand this? Set stable to 1 even commissioning data stable flag is zero. Or do you think that this is wrong?

jwhui, 2016-10-13:
I do not agree with that spec text.  There are actually a number of items incorrect.  For example, what exactly is the Operational Dataset TLV?  The Active or Pending version?

closed 2016-10-13 by lluwang