openthread/openthread #808: Spinel: Revisit handling of FCS in SPINEL_PROP_STREAM_RAW

darconeous, 2016-10-13:
I think we need to revisit how the FCS(Final CheckSum) is currently handled for `SPINEL_PROP_STREAM_RAW`.

Currently, The NCP *crops* the FCS from the packet if it isn't provided and indicates this condition with a flag in the metadata. If the radio actually supports FCS, then the FCS will only be present for inbound frames, not outbound frames. This means that outbound frames are actually two bytes shorter than inbound frames in that case.

`wpantund` handles this (and other things, like RSSI, etc) by using the [PPI DLT](http://www.cacetech.com/documents/PPI%20Header%20format%201.0.7.pdf), which allows you to both specify the DLT per packet as well as append additional metadata like RSSI, etc.

But I think I'd rather move to an approach where simply interpreting the packet and ignoring the metadata "just works". Sure, the CRCs will read as zero (and, thus, as an error) when they aren't set, but lots of sniffer implementations do that for outbound packets—and it's better than (effectively) corrupting the packet by interpreting the last two non-FCS bytes of the frame as an actual FCS.

darconeous, 2016-10-13:
Looking more closely, I think I totally botched FCS handling much more than I already thought I had. I'm preparing a pull request now.

jwhui, 2016-10-14:
Resolved by #809.

closed 2016-10-14 by jwhui