openthread/openthread #814: Hosted Doxygen Documentation out of date?

IOTUser, 2016-10-14:
According to your start page:
"The Doxygen reference docs are hosted online and generated as part of the build."
Leading to https://openthread.github.io/openthread/

But looking into files for e.g. platform abstraction shows many functions not documented on this website.
Are these functions undocumented or is something wrong with the automatic generation?

jwhui, 2016-10-14:
Unfortunately, while the docs are generated as part the build process, the docs are not automatically published as part of the continuous integration process.  We're working on a way to have the docs published automatically.  If you want the docs for a given commit ID, you can perform the standard bootstrap, configure, make process to generate the docs yourself.  You'll need doxygen for the basic doc support and graphviz if you want the call graphs as well.

jwhui, 2016-10-14:
BTW, I submitted #818 to update the hosted docs to ddd0704.

beriberikix, 2016-10-14:
What about looking into the doc publishing mentioned in #486?

jwhui, 2016-10-14:
Using master branch to host autogenerated docs does not seem ideal.  It would mean nearly every commit would require another commit to update the docs in master branch.  Also, the full set of generated docs is nearing 350 MB, which I think is a bit hefty to include in master.

beriberikix, 2016-10-14:
Good points! Didn't realize the docs feature had to be in master. #goodtoknow

jwhui, 2016-10-14:
Resolved by #818.

PR #818 (2016-10-14): Generated docs from ddd0704.
Resolves #814.
closed 2016-10-14 by jwhui