openthread/openthread #825: Failed to build OpenThread on CC2538 platform with the newer toolchain

chshu, 2016-10-17:
It fails when building CC2538 platform with a newer toolchain, for example: [5-2016-q2-update](https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update):

Build steps:
```
./bootstrap
make -f example/Makefile-cc2538
```

Build failure logs:

```
In file included from /home/chenshu/code/openthread/examples/../src/core/openthread.cpp:47:0:
/home/chenshu/code/openthread/examples/../src/core/common/new.hpp:42:15: error: 'void operator delete(void*, size_t)' is a usual (non-placement) deallocation function in C++14 (or with -fsized-deallocation) [-Werror=c++14-compat]
 void OT_CDECL operator delete(void *, size_t) throw();
               ^
In file included from /home/chenshu/code/openthread/examples/../src/core/net/ip6.cpp:38:0:
/home/chenshu/code/openthread/examples/../src/core/common/new.hpp:42:15: error: 'void operator delete(void*, size_t)' is a usual (non-placement) deallocation function in C++14 (or with -fsized-deallocation) [-Werror=c++14-compat]
 void OT_CDECL operator delete(void *, size_t) throw();
               ^
/home/chenshu/code/openthread/examples/../src/core/openthread.cpp:66:15: error: 'void operator delete(void*, size_t)' is a usual (non-placement) deallocation function in C++14 (or with -fsized-deallocation) [-Werror=c++14-compat]
 void OT_CDECL operator delete(void *, size_t) throw() { }
```

We could build CC2538 platform with an older toolchain, for example: [4.9-2015-q3-update](https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q3-update).

chshu, 2016-10-17:
@nibanks , Please help to take a look at the build issue, since it is involved with PR #767.

chshu, 2016-10-17:
@nibanks , It seems that we have another build issue that @wbober encountered when building with gcc 6.2.1: #826.

vinvinodk, 2016-10-17:
I still getting error while building on new toolchain with @wbober flag-addition tip. Below is the log.

Any inputs?
--------------------------------------------------------------------------------------------------------------------------------

Making all in apps
Making all in cli
  CCLD     ot-cli
../../../src/cli/libopenthread-cli.a(libopenthread_cli_a-cli_uart.o):(.rodata._ZTIN6Thread3Cli6ServerE[_ZTIN6Thread3Cli6ServerE]+0x0): undefined reference to `vtable for __cxxabiv1::__class_type_info'
../../../src/cli/libopenthread-cli.a(libopenthread_cli_a-cli_uart.o):(.rodata._ZTIN6Thread3Cli4UartE+0x0): undefined reference to `vtable for __cxxabiv1::__si_class_type_info'
/usr/bin/../lib/gcc/arm-none-eabi/5.4.1/armv7-m/libgcc.a(unwind-arm.o): In function `get_eit_entry':
unwind-arm.c:(.text+0x138): undefined reference to `__exidx_start'
unwind-arm.c:(.text+0x13c): undefined reference to `__exidx_end'
collect2: error: ld returned 1 exit status
Makefile:478: recipe for target 'ot-cli' failed
make[5]: *** [ot-cli] Error 1
Makefile:422: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
Makefile:424: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
Makefile:578: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
Makefile:503: recipe for target 'all' failed
make[1]: *** [all] Error 2
examples/Makefile-cc2538:235: recipe for target 'do-build-cortex-m3' failed
make: *** [do-build-cortex-m3] Error 2

chshu, 2016-10-18:
@vinvinodk , I have never seen this build error before, please verify your toolchain and follow the instructions in [README.md](https://github.com/openthread/openthread/blob/master/examples/platforms/cc2538/README.md).

vinvinodk, 2016-10-18:
@chshu sorry for that. i am just new when i was posting this, now i understand.

PR #832 (2016-10-18): Add arm-gcc54 to travis and -Wno-c++14-compat to CXXFLAGS.
Resolves #825.
closed 2016-10-18 by jwhui