openthread/openthread #828: Trying to run cli example on TI hardware

nirajthakre, 2016-10-17:
./bootstrap
autom4te: need GNU m4 1.4 or later: /cygdrive/c/openthread-master/third_party/nl                                                                                                      build-autotools/repo/tools/host/x86_64-unknown-cygwin/bin/m4
aclocal: error: echo failed with exit status: 1

I am getting this error on windows.
i have checked the version its same as mention in the error 
still it is giving the same error
please help me

jwhui, 2016-12-06:
@nirajthakre, I don't have a cygwin setup that I can test easily.  Did you end up resolving this issue?

LuDuda, 2016-12-06:
We have managed to build and run OpenThread using cygwin (32-bit version - since this is the only one pre-build in OpenThread's autotools).

One known issue is that arm-none-eabi-gcc does not recognize cygwin's path to the files, but it can be fixed by applying --srcdir option with Windows path to the configure:

e.g.:
`./configure [...] --srcdir="C:/Git/Openthread" --disable-docs`

The another issue that you may have is that you cannot have spaces in the arm-none-eabi toolchain path (By default they are installed in the "Program Files (x86)" which is invalid).

All in all we have been able to compile and run both posix and arm version, but it may be a paintful expirience to install and then to build (autotools works very slow) :-1:  so I recommend to start using native linux right away :+1:

nibanks, 2016-12-06:
I do all my work entirely on Windows, but when I need to build OpenThread for Linux I am using the new Windows 10 feature [Windows Subsystem for Linux](https://msdn.microsoft.com/en-us/commandline/wsl/about). If you have Windows 10, I'd suggest going that route.

closed 2016-12-14 by jwhui

nibanks, 2016-12-15:
As a follow up, I have put together a Wiki for explaining how to setup and build everything on Windows [here](https://github.com/openthread/openthread/wiki/Building-on-Windows).