openthread/openthread #842: MTU should be 1280 bytes

darconeous, 2016-10-18:
@nibanks brought to my attention that currently the MTU in OpenThread is being set to 1500 bytes. 

This value should be changed to 1280 bytes.

jwhui, 2016-10-19:
I agree 1500 bytes is not the proper value.  However, OpenThread needs to handle IPv6 datagrams slightly larger than 1280 bytes.  MPL multicast forwarding for destinations larger than Realm-Local scope requires IPv6-in-IPv6 encapsulation.  The 1280 byte MTU applies to the inner IPv6 datagram.  The outer IPv6 datagram would add another IPv6 header (40 bytes) and IPv6 hop-by-hop option header (8 bytes), so OpenThread needs to be able to handle up to 1328 byte IPv6 datagrams.

nibanks, 2016-10-19:
1. So, should OpenThread only allow packets of 1280 or less for calls to `otSendIp6Datagram`?
2. As long as the Thread spec allows for higher MTU, does it make sense to make the MTU a configurable value (for both OpenThread & Spinel)?

darconeous, 2016-10-19:
@jwhui wrote:

> The 1280 byte MTU applies to the inner IPv6 datagram. The outer IPv6 datagram would add another IPv6 header (40 bytes) and IPv6 hop-by-hop option header (8 bytes), so OpenThread needs to be able to handle up to 1328 byte IPv6 datagrams.

So then we would have two MTUs: The MTU exposed to applications (1280) and then an internal MTU that includes the wrapper (1328)...? 

> As long as the Thread spec allows for (see [1]) higher MTU, does it make sense to make the MTU a configurable value (for both OpenThread & Spinel)?

Larger MTUs have real consequences for embedded devices which want to participate on a Thread network. Even adding the option for an administratively configured MTU would require significant additional implementation complexity and resource usage.

jwhui, 2016-10-19:
> So then we would have two MTUs: The MTU exposed to applications (1280) and then an internal MTU that includes the wrapper (1328)...?

That makes sense to me.

PR #1032 (2016-12-01): Reduce MTU to 1280 bytes
Fixes #842.
closed 2016-12-01 by jwhui