openthread/openthread #853: NCP Doesn't Fully Expose the OpenThread API

nibanks, 2016-10-20:
I have been adding Spinel/NCP support to my Windows driver, to support running OpenThread directly on device, and I have found that a number of the OpenThread APIs aren't exposed via the Spinel/NCP interface. The following are the APIs that have no support:

1. otAddMacBlacklist
2. otClearMacBlacklist
3. otCommissionerAddJoiner
4. otCommissionerAnnounceBegin
5. otCommissionerEnergyScan
6. otCommissionerPanIdQuery
7. otCommissionerProvisioningUrl
8. otCommissionerRemoveJoiner
9. otCommissionerStart
10. otCommissionerStop
11. otDisableMacBlacklist
12. otDiscover
13. otEnableMacBlacklist
14. otGetActiveDataset
15. otGetAssignLinkQuality
16. otGetEidCacheEntry
17. otGetFactoryAssignedIeeeEui64
18. otGetHashMacAddress
19. otGetJoinerUdpPort
20. otGetLeaderData
21. otGetLocalLeaderPartitionId
22. otGetMaxAllowedChildren
23. otGetMaxTransmitPower
24. otGetNetworkDataLeader
25. otGetPendingDataset
26. otGetPollPeriod
27. otGetRouterIdSequence
28. otGetRouterInfo
29. otIsSingleton
30. otJoinerStart
31. otJoinerStop
32. otMacBlacklistEntry
33. otRemoveMacBlacklist
34. otSendActiveGet
35. otSendActiveSet
36. otSendDiagnosticGet
37. otSendDiagnosticReset
38. otSendMgmtCommissionerGet
39. otSendMgmtCommissionerSet
40. otSendPendingGet
41. otSendPendingSet
42. otSetActiveDataset
43. otSetAssignLinkQuality
44. otSetJoinerUdpPort
45. otSetLocalLeaderPartitionId
46. otSetMaxAllowedChildren
47. otSetMaxTransmitPower
48. otSetPendingDataset
49. otSetPollPeriod

otSendServerData is exposed in only a partial way, and I am confused at exactly how the interface around that API is supposed to work. I'd ask/suggest that we either change or add a command to directly call this API.

Currently, there is no way to query the entire output of otMacCounters. I'd suggest we add a command for that.

jwhui, 2016-10-20:
I get that you are trying to maintain a Windows driver that can interface with both the OpenThread API and NCP (as you mentioned in #855).  However, I'm not sure the Windows driver API needs to map 1:1 with OpenThread APIs.

Some questions:

1. I noted that you did not list the (recently added) CoAP APIs, was that intentional or do you plan to include Windows driver support for those as well?
2. Do you intend to operate the Commissioner role on the NCP?  Due to Commissioner resource requirements, I would expect the Commissioning role to be implemented on the host processor or some external device (i.e. mobile).  The OpenThread implementation, for example, does not support for app-specific commissioning over the DTLS session.
3. Some of the APIs only exist to support the Thread Certification Harness Reference Device role, which requires a mapping through the [Thread Harness THCI](https://github.com/openthread/openthread/tree/master/tools/harness-thci).  Are you looking to support this in Windows as well?  If you are only operating as a Device Under Test, these APIs are not necessary.

nibanks, 2016-10-20:
Eventually, on Windows, everything external to the core Thread logic, will sit on top of the otApi.dll. This DLL currently exposes nearly every control path API exposed by openthread.h. It purposefully excludes any API that would be socket related, as that is expected and required to use the normal Windows Socket APIs. otApi.dll will implement each control path API and communicate it to the otLwf.sys filter driver via IOCTLs. Depending on the currently connect device, otLwf.sys will then either call OpenThread APIs directly in response to the IOCTL, or encode it with spinel to send directly to the device.

All user experiences and testing go through otApi.dll. The Thread Harness for Windows will have to go through it. Any off-mesh commissioning logic will go through it. On-mesh commissioning logic is considered as part of the core Thread logic, so currently, would be expected to live on device if that is where OpenThread is running.

jwhui, 2016-10-27:
I'm fine with exposing all of the control path parameters.  The commissioner role blurs the line since additional application-specific provisioning could happen through the DTLS session.  We have not exposed an API that exposes this path.  Would you consider this control or data path?

nibanks, 2016-10-27:
I'm pretty sure I'd still consider it control path, though I agree it blurs the lines. I am fine with holding off commissioning related APIs to NCP until they have been finalized.

jwhui, 2017-10-02:
Closing stale issue.

closed 2017-10-02 by jwhui