openthread/openthread #855: NCP Limited otBecomeChild Support

nibanks, 2016-10-20:
NCP doesn't support setting the second parameter of `otBecomeChild` to anything other than `kMleAttachAnyPartition`.

jwhui, 2016-10-20:
It's worth noting that some API functionality was exposed simply to assist with early development.  I'd argue that functions such as `otBecomeDetached()` and `otBecomeChild()` should actually be removed from the API.  Detaching can be achieved by calling `otThreadStop()`.  Becoming a child can be achieved by calling `otSetRouterRoleEnabled(false)`.  We no longer use `otBecomeDetached()` and `otBecomeChild()` in our testing.  I've been meaning to do a cleanup pass on the API.

nibanks, 2016-10-20:
That makes sense. I'm fine either way, as long as we keep NCP and the API in sync. Because I expose the API and a DLL on Windows, and the driver needs to support both calling the API directly, and using NCP to invoke it on the device, I need them in sync.

mszczodrak, 2016-10-20:
@nibanks , your driver's design sounds interesting. Do you plan to open-source your driver's design? Is it made just for Windows or is it portable driver?

nibanks, 2016-10-20:
It will only work on Windows, and yes, it will be open sourced. I have a PR out for the first stage of the driver that supports running OpenThread in Windows, and I am actively working on adding support for NCP on one of [my branches](https://github.com/nibanks/openthread/tree/win/spinel). The [Filter driver](https://msdn.microsoft.com/en-us/windows/hardware/drivers/network/ndis-filter-drivers) part is almost done, but I still have to write the [miniport](https://msdn.microsoft.com/en-us/windows/hardware/drivers/network/ndis-miniport-drivers2) to actually talk to the device.

jwhui, 2017-05-24:
The attach mode argument was removed from the API as a part of #1816.

closed 2017-05-24 by jwhui