openthread/openthread #870: Analysis: Thread1.1 9.2.17-Orphaned End Device

xiaom-GitHub, 2016-10-23:
This case requires two sniffers to listen on different channels. Originally I set the channel to 12 and 13, the traffic looks normal as expected with PR #869 . However, Harness reports it as failure. The result as below:
[ED_9_2_17_Channel_12_13.zip](https://github.com/openthread/openthread/files/546313/ED_9_2_17_Channel_12_13.zip)

From pcap and putty log, we can see all packets that Harness would verify are existing. I guess that fail may be due to it must operate on channel 18 (according to test plan, it specifies that Leader_2 needs to run on channel 18). Hence, I re-run this case on channel 12 and 18. But on channel 12 and 18, the device's behavior seems not be normal. 
[ED_9_2_17_Channel_12_18.zip](https://github.com/openthread/openthread/files/546315/ED_9_2_17_Channel_12_18.zip)

I also attach the putty log that connecting with TI CC2538 board which enables logging. So that we can see the traffic exchanges. When operating on channel 12 and 18, end device on channel 12 will send out MLE Announce on channel 18 after it loses the connectivity with its parent. Leader on channel 18 will also response and send MLE Announce on channel 12. But end device seems not receive this announce message, that cause reattach fails.

Then I change to channel 12 and 16, the same issue occurs.
[ED_9_2_17_Channel_12_16.zip](https://github.com/openthread/openthread/files/546321/ED_9_2_17_Channel_12_16.zip)

But when I verify with posix nodes, everything goes well and reattach successes. @jwhui Could you please help take a look at this? Thanks much.

jwhui, 2016-10-24:
I'm not exactly sure what the issue is here.  This may be due to a channel configuration issue in the CC2538 radio driver.  The current driver does not disable the receiver before configuring the channel.  I just submitted #873 to address this issue.  Let me know if this helps.

xiaom-GitHub, 2016-10-25:
@jwhui Thanks much for addressing this issue, PR #873 works well. I verify this with Harness and simulate manually, both traffic looks normal as expected.
[ED_9_2_17.zip](https://github.com/openthread/openthread/files/549603/ED_9_2_17.zip)

However, test case 9.2.17 is designed incompatible with Spec(should delay 8 hours and then new panid will take effect).I have filed a JIRA [TSTBED-25](https://threadgroup.atlassian.net/browse/TSTBED-25) to have a confirm with GRL.

PR #873 (2016-10-25): Disable receiver when changing the channel config.
Potentially fixes #870.
closed 2016-10-25 by jwhui