openthread/openthread #891: API: Multicast address subscribe/unsubscribe

LuDuda, 2016-10-27:
Currently OpenThread API does not allow to add / remove IPv6 address of multicast type. Do we want to have such possibility for example to be able to add such address in the CLI example? If yes, should we reuse ipaddr command and detect whether address is of multicast type, or we want to have separate commands for it?

nibanks, 2016-10-27:
For Windows, I'd prefer to add the option to get forwarded all multicast messages, since the Windows networking stack will take care of the rest.

jwhui, 2016-10-27:
I agree that the API should include multicast address subscribe/unsubscribe.  I also agree that the API should allow passing all received IPv6 multicast messages to the host without explicitly subscribing.  @LuDuda, are you willing to submit a PR for these enhancements?

LuDuda, 2016-10-27:
Yes, you can assign me for that.

PR #899 (2016-11-02): IP: Add API to subscribe/unsubscribe to multicast addresses.
This PR fixes #891 issue by adding external API to subscribe and unsubscribe multicast addresses.
Also i have added possibility of setting the multicast promiscuous mode suggested by @nibanks to accept all, where for example we act as an NCP.

There were no external registration for multicast addresses, so i mostly reused same code as it is for unicast addresses.

I follow numenclature of Linux and `ip maddr` command, so the CLI command is called `ipmaddr`.

The only problematic case is with command that list multicast addresses. Cur [... cut]
closed 2016-11-02 by jwhui