openthread/openthread #910: Analysis: Thread1.1 9.2.9-Sync Pending Commissioning Datasets when 2 partitions merge

xiaom-GitHub, 2016-11-01:
Currently there are two problems about OpenThread implementation that would affect this case:
1. Child ID Response message does not include Active Timestamp Tlv if child has a same value with its parent. It's compatible with spec requirement. However, harness will check this filed in Child ID Response massage. 
(If Active Timestamp Tlv and Pending Timestamp Tlv would not bring negative impact, from testing purpose, maybe we can add them back)
 
2. For now, Pending Dataset will only include TLVs carried from MGMT_PENDING_SET command. However, Harness will check whether or not it also contains those TLVs in Active Dataset as well such as security policy tlv, PSKc tlv, channelMask tlv.... (during initialization Harness configures these Tlvs via THCI, so we can see that Active Dataset includes all TLVs.)

Spec states that
"The Pending Operational Dataset contains all the parameters from the Active Operational Dataset, with the addition of Delay timer and pending timestamp".
Maybe we can append these TLVs from Active Dataset to Pending Dataset as well. Please let me know your advice. Thanks.

jwhui, 2016-11-01:
> 1. Child ID Response message does not include Active Timestamp Tlv if child has a same value with its parent. It's compatible with spec requirement. However, harness will check this filed in Child [...]

I'm fine with adding the Active and Pending Timestamp TLVs back to the Child ID Response message.  Let's also file a JIRA to raise the issue to the Thread Group.

> 2. For now, Pending Dataset will only include TLVs carried from MGMT_PENDING_SET command. However, Harness will check whether or not it also contains those TLVs in Active Dataset as well such as [...]

If the MGMT_PENDING_SET command is from a Commissioner, then the Leader should first initialize its Pending Dataset with the Active Dataset TLVs, then apply the TLVs received in the MGMT_PENDING SET message.  If the MGMT_PENDING_SET command is not from a Commissioner, then the copy should not occur (the copy should have already occurred before the device received the Pending Dataset).

Make sense?

xiaom-GitHub, 2016-11-01:
Thanks for this advice. For 2, i understand that if MGMT_PENDING_SET from a thread device, it has already synced pending dataset within network, at this moment, when it sends MGMT_PENDING_SET command with a newer pending timestamp,  Leader should not need to copy it again, just apply the changes in pending dataset.
However, if no commissioner sends MGMT_PENDING_SET to leader first, in this case, do we need to process this copy when Leader receives MGMT_PENDING_SET from a thread device?

jwhui, 2016-11-01:
A non-Commissioner needs to always send the Pending Dataset as a complete dataset (i.e. assumes that the Leader will not copy TLVs from the Active Dataset).  If a Thread device only wants to update selected TLVs, then it needs to operate as a Commissioner and send the MGMT_PENDING_SET to the Leader, so that the Leader can fill in the rest of the (unchanged) TLVs.

xiaom-GitHub, 2016-11-01:
Got it, that makes sense. Will submit a PR to address this. Thanks.

xiaom-GitHub, 2016-11-03:
close this issue via #915 .

closed 2016-11-03 by xiaom-GitHub