openthread/openthread #918: Include local Dataset's Timestamp TLV in Child ID Request message after reset

chshu, 2016-11-02:
This is the last issue to pass Certification test 9.2.15 and 9.2.16([Router_9_2_15.zip](https://github.com/openthread/openthread/files/566649/Router_9_2_15.zip)), the test steps:
1. DUT attaches network and store Active Dataset and Pending Dataset locally;
2. DUT power down for a moment, some changes will happen to the network;
3. DUT power on and try to reattach to network, Harness checks the present of original Active/Pending Timestamp TLVs in Child ID Request message, we failed with this check.

But I think it 's better to **not** include local stored Dataset's Timestamp TLV in Child ID Request message, since after the reset (assume no changes happened to the network), the Delay Timer should be out of date, and we need to get the refreshed Pending Dataset with updated Delay Time from Child ID Response message. But if we include the local Timestamp TLVs in Child ID Request message as Harness expected, the parent will not include Datasets in Child ID Response. @jwhui , May I know your opinion on this issue? Thanks.

However, we could still make some changes to satisfy Harness check and pass the tests:
Include the local Dataset's Timestamp TLV in Child ID Request message, and enable local Datasets as network Datasets if parent do not indicate that it has a different Datasets.

jwhui, 2016-11-02:
I agree with you that we should not include the Pending Timestamp TLV in the Child ID Request.  Please open a JIRA to raise this question with the Thread Group.

Let's add the workaround for now and put a comment there to say that we probably should not including the Pending Timestamp TLV.

chshu, 2016-11-03:
I have filed the JIRA to Thread Group for this issue: [SPEC-668](https://threadgroup.atlassian.net/browse/SPEC-668).
And also have a workaround to pass test 9.2.15 and 9.2.16: #922.

PR #922 (2016-11-03): Include local Dataset Timestamp TLVs in Child ID Request message after reset
Resolves #918.

It helps to pass Certification test 9.2.15 and 9.2.16, will upload logs soon.
closed 2016-11-03 by jwhui