openthread/openthread #933: Test Cert_8_2_01_JoinerRouter.py seems Unreliable

nibanks, 2016-11-04:
I have had several failures that required a rerun to resolve, failing because of `Cert_8_2_01_JoinerRouter.py`, [for example](https://travis-ci.org/nibanks/openthread/jobs/173293903).

I see the following error in the console output:

```
2016-11-04 16:19:19.413854 CRIT MCOP mbedtls_ssl_read_record() returned -30848 (-0x7880)
```

This seems to map to:

```
/**< The peer notified us that the connection is going to be closed. */
#define MBEDTLS_ERR_SSL_PEER_CLOSE_NOTIFY                 -0x7880 
```

jwhui, 2016-11-04:
I've seen this too.  I haven't had a chance to root cause this yet.

nibanks, 2016-11-04:
As far as I can tell, this will only happen if the other side calls `mbedtls_ssl_close_notify`. That seems to be called in two places:

- `Dtls::Process` in response to receiving a result of `MBEDTLS_ERR_SSL_PEER_CLOSE_NOTIFY`, obviously not the culprit here.
- `Dtls::Stop` which must be the issue here. That is called from `Dtls::HandleMbedtlsSetTimer` if `aFinish == 0`. I'm not exactly sure what that means. Did something timeout?

chshu, 2016-11-05:
The CLOSE_NOTIFY is expected steps by the end of DTLS handshake which is also called by `Joiner::Close()` in `joiner.cpp`, so it maybe not the root cause, I will take a look at this issue.

nibanks, 2016-11-16:
Now I just hit a similar error on `Cert_8_1_01_Commissioning' ([build link](https://travis-ci.org/nibanks/openthread/jobs/176504186)):

```
2016-11-16 21:16:56.211905 CRIT MCOP is a fatal alert message (msg 40)
2016-11-16 21:16:56.212013 CRIT MCOP mbedtls_ssl_read_record() returned -30592 (-0x7780)
```

chshu, 2016-11-17:
I tried but failed to reproduce the failure locally with many runs, may need more efforts to locate the root cause.

PR #1970 (2017-07-11): [mbedtls] implement heap for mbedtls
This PR implements a new memory allocator for mbedTLS. It fixes #933 and saves about 2k RAM.

DTLS related tests fail occasionally. The direct cause is failing to allocate memory when doing ECP points operations. OpenThread provides enough memory for mbedTLS, however, whether it fails depends on the situation of memory fragmentation. When too many memory wasted, it fails for being unable to find continuous memory for new allocations.

Besides, under 32bit system mbedTLS uses 32 bytes(it doubles on 64bit system) metadata for each allocations. To establish a  [... cut]
closed 2017-07-11 by jwhui