openthread/openthread #950: Test Cert_9_2_11_MasterKey is Unreliable

nibanks, 2016-11-08:
I am getting failures in Cert_9_2_11_MasterKey.py fairly consistently on my master branch, which I keep as a clean clone of openthread/master. See failure [here](https://travis-ci.org/nibanks/openthread/jobs/174272023).

```
======================================================================
FAIL: test (__main__.Cert_9_2_11_MasterKey)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/travis/build/nibanks/openthread/examples/../tests/scripts/thread-cert/Cert_9_2_11_MasterKey.py", line 150, in test
    self.assertEqual(self.nodes[SED1].get_masterkey(), KEY1)
AssertionError: u'0f0e0d0c0b0a09080706050403020100' != '000102030405060708090a0b0c0d0e0f'

----------------------------------------------------------------------
Ran 1 test in 58.407s
```

jwhui, 2016-11-08:
#912 should address this issue.

nibanks, 2016-11-08:
I just rebased this morning, and hit this failure two build tries in a row. #912 is already merged into master, right?

jwhui, 2016-11-08:
Oops, I meant #946.  Updated above comment to reflect that.

PR #946 (2016-11-08): Schedule MLE Data Response messages to SEDs on network data change.
This addresses the delivery delay raised in #912.

This also addresses some of the spurious Travis failures caused by delays in delivering the Data Response message to SEDs (e.g. 9.2.11).

Resolves #950.
closed 2016-11-08 by jwhui