openthread/openthread #960: [spinel-py] Remove TunInterface depedency from codec.py

wbober, 2016-11-09:
Here's the use case: I'd like to use the spinel package on Windows to control an NCP. To make this possible I'd like to:

- Remove the dependency on TunInterface from codec.py. It looks to me that it's only there to be accessible for `if_up` and `if_down` methods. These are called from `spinel-cli.py`. How about making the TunInterface a member object of `SpinelCliCmd` class and removing self.tun_if member from `WpanApi` class.
- A side effect of the change above is that we can get rid of nodeid from WpanApi constructor. Which, I think, is desired since nodeId is simulator specific whereas the WpanApi isn't
- All code which refers to the TunInterface would be guarded to be available only on Posix.

Are there any objections to these changes?

jwhui, 2016-11-09:
I'm always an advocate of removing dependencies and simplifying as much as possible.  Is TunInterface even worthwhile to keep around?

turon, 2016-11-09:
I agree that codec.py should not have a dependency on tun.py.  I also agree that removing the need to track nodeid from spinel-cli is worthwhile.  That said, I think the TunInterface remains a valuable feature for Linux and Mac, and I'd support moving it to only be included by spinel-cli.py.

jwhui, 2016-11-09:
@turon, can you elaborate more on the value of TunInterface relative to wpantund?

turon, 2016-11-09:
@jwhui Agree that TunInterface is starts creating redundancy to the wpantund feature set.
To the extent that spinel-cli provides a good testing and fast bring-up tool for python users, I'd say the value is:
1) Provides a means to use standard python sockets for python-based test apps.
2) Provides a means to use stock posix IPv6 apps such as iperf, curl, dig, etc along side the familiar ot-cli management shell.

jwhui, 2016-11-10:
@turon wpantund also allows standard IPv6 performance measurement tools and socket-based programming.  I think it's better to include wpantund in these kinds of system-level tests.  The downside of TunInterface is that it relies on the host's IPv6 stack, which may limit the kinds of packets that you can inject into or receive from the Thread network and the kinds of tests that you can perform.

wbober, 2016-12-05:
Fixed by #1041.

closed 2016-12-05 by wbober