openthread/openthread #961: [spinel-py] A separate repository for spinel package

wbober, 2016-11-09:
How to you feel about making a separate repository for the spinel package? I think the code might be useful for writing simple PC tools which need to connect to the Thread network through the NCP.

A separate repository would make the code more visible to potential users and make the commit history more clean.

wbober, 2016-11-23:
@turon Any thoughts?

nibanks, 2016-11-23:
Is there a reason you don't want to just put the code for the tools in this repository?

turon, 2016-11-29:
@wbober sorry for the slow response.  I don't have a strong opinion, but my leaning is to leave it in the main openthread repo.  One nice thing about it being part of the main repo is that it is tightly integrated into the continuous integration, which was the original driving goal for the package.

wbober, 2016-12-05:
@nibanks Not really. I just thought it makes it more visible.
@turon  N/p

Conclusion: let's leave it as is. Reconsider if anyone else asks.

wbober, 2016-12-19:
I've experimented with adding dependency on spinel package directly from git. The drawback is that the whole repository, which is quite heavy, must be cloned in order to install the package.

Could we please reconsider a separate repository for spinel? An alternative is to place spinel package in pypi but I don't think it's the best thing to do at this stage.

@turon  Would using the package as a submodule work for the continuous integration scenario?

turon, 2016-12-19:
@wbober Submodule works for me.  Would it just go straight into the same tools directory it lives in now?

wbober, 2016-12-19:
@turon Yes, I expect so. It would probably require some adjustments to the CI scripts to pull the submodule.

wbober, 2017-04-06:
@turon Can we put this into action?

I think we need to do the following:
1) Create `openthread/spinel-cli` repository (ot team)
2) Move spinel-cli code from openthread to the new repository (me)
3) Add spinel-cli as a submodule to openthread (me)
4) Modify CI scripts to pull the submodule (not yet sure who can do that)

turon, 2017-04-06:
@wbober sure, I can discuss the first step with @jwhui tomorrow morning and help with step 4.

wbober, 2017-04-06:
@turon Great.

jwhui, 2017-04-06:
@wbober, I'm fine with separating spinel tools into a separate repo.  If we view spinel-cli as a separate package, is there a real need to include the spinel-cli repo as a git submodule within OpenThread?  We could just have the Travis scripts install the spinel-cli package as part of the setup process, which should be trivial.

wbober, 2017-04-07:
@jwhui Yes, you are right.

jwhui, 2017-04-07:
What about naming the package and repo `pyspinel`?  Open to other suggestions as well.

wbober, 2017-04-07:
Yes, that's a good idea.

jwhui, 2017-04-10:
Any concerns with using Apache 2 on the new pyserial repo?

wbober, 2017-04-20:
@jwhui 

Sorry for the delay - I was on holidays. 

No concerns with Apache 2.

PR #1742 (2017-05-10): spinel-cli is now maintained in a separate repo.
Resolves #961.
closed 2017-05-10 by jwhui