openthread/openthread #967: Thread device keeps registering newer dataset to Leader during dataset transition

xiaom-GitHub, 2016-11-12:
Observe the phenomenon, for example, there are two nodes, one is Leader and another is Router. Leader and Router forms a partition respectively (Leader has a larger partition id than Router's). Configure an older active timestamp on Leader's partition (e.g. 10) and a newer active timestamp (e.g. 20) on Router's.
Then remove the whitelist and let Router can hear Leader's advertisement. Router will migrate to Leader's partition. And when receiving the Child Id Response, it will register to Leader with its newer active dataset. However, there are also Tlvs that would affect network connectivity included in this MGMT_ACTIVE_SET.req, so Leader moves the entire Active dataset to Pending dataset and starts the delay timer. Due to the new active timestamp does not take effect immediately on Leader's side. Router does not get the chance to update its `mNetwork` with `mLocal` version, it will keep registering the its own dataset to Leader until the delay timer expires.

xiaom-GitHub, 2016-11-12:
To address this, maybe we can relay on whether or not receiving the MGMT_ACTIVE_SET.rsp or MGMT_PENDING_SET.rsp. We can set a flag as `false` when sending MGMT_ACTIVE_SET.req or MGMT_PENDING_SET.req, and then set this flag as `true` when receiving the response with `Accept` state TLV. If this flag is `true`, we stop registering since peer has received the request message, and if it's still `false`, we keep registering until the response is received. Is this doable?

jwhui, 2016-11-14:
I think stopping registration attempts after receiving a MGMT_ACTIVE_SET.rsp message will handle most cases.  However, I think it would be more robust to check that the Pending Dataset received from the network includes the Active Dataset's Timestamp.  In the corner case where the Pending Dataset has an even newer Active Timestamp, then there is nothing the device can do and should not even register in the first place.  In other words, if the Pending Dataset includes an Active Timestamp >= to the Active Dataset Timestamp, the device should not attempt to send a MGMT_ACTIVE_SET.req.  Thoughts?

xiaom-GitHub, 2016-11-14:
Yes, checking the active timestamp in pending dataset received from network makes sense. Will submit a PR for this later. Thanks.

xiaom-GitHub, 2016-12-13:
closed this issue via #1061 .

closed 2016-12-13 by xiaom-GitHub