openthread/openthread #977: Gcc version detection

blauret, 2016-11-15:
GitHub Issues are for bugs and feature requests.  To make bugs and feature requests more easy to find and organize, we close issues that are deemed out of scope for GitHub Issues.

Usage questions? Post questions to [Stack Overflow](http://stackoverflow.com/) using the [[openthread] tag](http://stackoverflow.com/questions/tagged/openthread). We also use Google Groups for discussion and announcements:

* [openthread-announce](https://groups.google.com/forum/#!forum/openthread-announce) - subscribe for release notes and new updates on OpenThread
* [openthread-users](https://groups.google.com/forum/#!forum/openthread-users) - the best place for users to discuss OpenThread and interact with the OpenThread team
* [openthread-devel](https://groups.google.com/forum/#!forum/openthread-devel) - team members discuss the on-going development of OpenThread

blauret, 2016-11-15:
My experience is Thread does not compile with gcc version 4.x. 

I saw somewhere in the build tools some checks about presence of gcc in the path. If OpenThread supports only gcc 5.x or greated, would it be possible to implement a check the version of gcc and display a nice message if it's too old?

jwhui, 2016-11-15:
OpenThread supports gcc 4, 5, and 6 and you can see the build results [here](https://travis-ci.org/openthread/openthread).

Can you provide your console output that includes the command line and error?

blauret, 2016-11-15:
Bens-mbpr:openthread ben$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.7.4 20130913 (release) [ARM/embedded-4_7-branch revision 202601]
Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Bens-mbpr:openthread ben$ make -f examples/Makefile-cc2538 
  MKDIR    output
  MKDIR    build/cortex-m3-arm-none-eabi
  CONFIG   cortex-m3-arm-none-eabi...
(cd build/cortex-m3-arm-none-eabi && /Users/ben/git/openthread/examples/../configure INSTALL="/usr/bin/install -p" CPP="arm-none-eabi-cpp" CC="arm-none-eabi-gcc" CXX="arm-none-eabi-g++" OBJC="" OBJCXX="" AR="arm-none-eabi-ar" RANLIB="arm-none-eabi-ranlib" NM="arm-none-eabi-nm" STRIP="arm-none-eabi-strip" CPPFLAGS="-fdata-sections -ffunction-sections -Os -g  -mcpu=cortex-m3 -mfloat-abi=soft -mthumb " CFLAGS="-fdata-sections -ffunction-sections -Os -g  -mcpu=cortex-m3 -mfloat-abi=soft -mthumb " CXXFLAGS="-fdata-sections -ffunction-sections -Os -g  -mcpu=cortex-m3 -mfloat-abi=soft -mthumb -fno-exceptions -fno-rtti " LDFLAGS="-fdata-sections -ffunction-sections -Os -g  -mcpu=cortex-m3 -mfloat-abi=soft -mthumb -nostartfiles -specs=nano.specs -specs=nosys.specs -Wl,--gc-sections -Wl,-Map=map.map " --host=arm-none-eabi --target=arm-none-eabi --prefix=/ --enable-cli --enable-diag --with-examples=cc2538 --with-platform-info=CC2538 )
configure: WARNING: if you wanted to set the --build type, don't use --host.
    If a cross compiler is detected then cross compile mode will be used
checking build system type... i386-apple-darwin15.6.0
checking host system type... arm-none-eabi
checking target system type... arm-none-eabi
checking filtered build system type... i386-apple-darwin
checking filtered host system type... arm-none-eabi
checking filtered target system type... arm-none-eabi
checking for a BSD-compatible install... /usr/bin/install -p
checking whether build environment is sane... yes
checking for arm-none-eabi-strip... arm-none-eabi-strip
checking for a thread-safe mkdir -p... /Users/ben/git/openthread/examples/../third_party/nlbuild-autotools/repo/autoconf/install-sh -c -d
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking whether make supports nested variables... (cached) yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking whether to disable executable checking... no
checking for style of include used by make... GNU
checking for arm-none-eabi-gcc... arm-none-eabi-gcc
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether arm-none-eabi-gcc accepts -g... yes
checking for arm-none-eabi-gcc option to accept ISO C89... none needed
checking whether arm-none-eabi-gcc understands -c and -o together... yes
checking dependency style of arm-none-eabi-gcc... gcc3
checking how to run the C preprocessor... arm-none-eabi-cpp
checking for arm-none-eabi-gcc... (cached) arm-none-eabi-gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether arm-none-eabi-gcc accepts -g... (cached) yes
checking for arm-none-eabi-gcc option to accept ISO C89... (cached) none needed
checking whether arm-none-eabi-gcc understands -c and -o together... (cached) yes
checking dependency style of arm-none-eabi-gcc... (cached) gcc3
checking whether arm-none-eabi-gcc and cc understand -c and -o together... yes
checking whether we are using the GNU C++ compiler... yes
checking whether arm-none-eabi-g++ accepts -g... yes
checking dependency style of arm-none-eabi-g++... gcc3
checking how to run the C++ preprocessor... arm-none-eabi-g++ -E
checking whether we are using the GNU C++ compiler... (cached) yes
checking whether arm-none-eabi-g++ accepts -g... (cached) yes
checking dependency style of arm-none-eabi-g++... (cached) gcc3
checking whether arm-none-eabi-g++ understands -c and -o together... yes
checking for arm-none-eabi-ar... arm-none-eabi-ar
checking for arm-none-eabi-ranlib... arm-none-eabi-ranlib
checking for arm-none-eabi-objcopy... arm-none-eabi-objcopy
checking for arm-none-eabi-strip... (cached) arm-none-eabi-strip
checking whether ln -s works... yes
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking whether arm-none-eabi-gcc is Clang... no
checking whether the C compiler understands -Wall... yes
checking whether the C compiler understands -Wextra... yes
checking whether the C compiler understands -Wshadow... yes
[... 308 lines cut ...]
  CC       libopenthread_cc2538_a-flash.o
  CC       libopenthread_cc2538_a-alarm.o
  CC       libopenthread_cc2538_a-logging.o
  CC       libopenthread_cc2538_a-misc.o
  CC       libopenthread_cc2538_a-platform.o
  CC       libopenthread_cc2538_a-radio.o
  CC       libopenthread_cc2538_a-random.o
  CC       libopenthread_cc2538_a-startup-gcc.o
cp /Users/ben/git/openthread/examples/../examples/platforms/cc2538/../utils/settings.cpp libopenthread_cc2538_settings.cpp
  CC       libopenthread_cc2538_a-uart.o
  CC       libopenthread_cc2538_a-diag.o
  CXX      libopenthread_cc2538_a-libopenthread_cc2538_settings.o
  AR       libopenthread-cc2538.a
make[5]: Nothing to be done for `all-am'.
Making all in apps
Making all in cli
  CC       ot_cli-main.o
  CCLD     ot-cli
arm-none-eabi-gcc: error: nosys.specs: No such file or directory
make[5]: *** [ot-cli] Error 1
make[4]: *** [all-recursive] Error 1
make[3]: *** [all-recursive] Error 1
make[2]: *** [all-recursive] Error 1
make[1]: *** [all] Error 2
make: *** [do-build-cortex-m3] Error 2

jwhui, 2016-11-15:
This does not seem to be a problem with the compiler, but a problem with the lib included with the compiler itself.  The distribution of arm-none-eabi-gcc that you are using does not include the nosys .specs file in its lib.

For example, from gcc-arm-none-eabi-4_9-2015q3:

```bash
$ find . -name "*nosys*"
./arm-none-eabi/lib/armv6-m/libnosys.a
./arm-none-eabi/lib/armv6-m/nosys.specs
./arm-none-eabi/lib/armv7-ar/thumb/fpu/libnosys.a
./arm-none-eabi/lib/armv7-ar/thumb/fpu/nosys.specs
./arm-none-eabi/lib/armv7-ar/thumb/libnosys.a
./arm-none-eabi/lib/armv7-ar/thumb/nosys.specs
./arm-none-eabi/lib/armv7-ar/thumb/softfp/libnosys.a
./arm-none-eabi/lib/armv7-ar/thumb/softfp/nosys.specs
./arm-none-eabi/lib/armv7-m/libnosys.a
./arm-none-eabi/lib/armv7-m/nosys.specs
./arm-none-eabi/lib/armv7e-m/fpu/libnosys.a
./arm-none-eabi/lib/armv7e-m/fpu/nosys.specs
./arm-none-eabi/lib/armv7e-m/libnosys.a
./arm-none-eabi/lib/armv7e-m/nosys.specs
./arm-none-eabi/lib/armv7e-m/softfp/libnosys.a
./arm-none-eabi/lib/armv7e-m/softfp/nosys.specs
./arm-none-eabi/lib/cortex-m7/fpu/fpv5-d16/libnosys.a
./arm-none-eabi/lib/cortex-m7/fpu/fpv5-d16/nosys.specs
./arm-none-eabi/lib/cortex-m7/fpu/fpv5-sp-d16/libnosys.a
./arm-none-eabi/lib/cortex-m7/fpu/fpv5-sp-d16/nosys.specs
./arm-none-eabi/lib/cortex-m7/libnosys.a
./arm-none-eabi/lib/cortex-m7/nosys.specs
./arm-none-eabi/lib/cortex-m7/softfp/fpv5-d16/libnosys.a
./arm-none-eabi/lib/cortex-m7/softfp/fpv5-d16/nosys.specs
./arm-none-eabi/lib/cortex-m7/softfp/fpv5-sp-d16/libnosys.a
./arm-none-eabi/lib/cortex-m7/softfp/fpv5-sp-d16/nosys.specs
./arm-none-eabi/lib/fpu/libnosys.a
./arm-none-eabi/lib/fpu/nosys.specs
./arm-none-eabi/lib/libnosys.a
./arm-none-eabi/lib/nosys.specs
./arm-none-eabi/lib/thumb/libnosys.a
./arm-none-eabi/lib/thumb/nosys.specs
```

blauret, 2016-11-15:
Hi Jonathan,

Thanks for the tip. I'll look closer into my toolchain

closed 2016-11-15 by blauret