openthread/openthread #1003: Processing advertisments on parent migration

wbober, 2016-11-22:
Looking at the current implementation I see that when a child receives
an advertisement from its parent, which indicates that the parent has changed
its partition, it updates the leader information. However, it looks to me that the 
Router Id Sequence number isn't updated.

This is problematic, since it might happen that the Router Id Sequence in 
the new partition is lower that in the former one. This means that the routing
information sent in consecutive advertisements won't be processed. If the child
is a REED then this will cause the node to think that it doesn't have connectivity
to the leader and in effect leave the partition.

Is that an intended behavior?

jwhui, 2016-11-22:
I also could not find how the Router ID Sequence value gets update as a result of the parent changing partitions.  Would you like to work on a fix?

rlubos, 2016-11-23:
I will look into this issue.

wbober, 2016-12-12:
Fixed in #1044

closed 2016-12-12 by wbober