openthread/openthread #1036: Some Test cases failed with 'sim' on Ubuntu.

bot-zjlu, 2016-12-02:
Hi,

We tried to run the Cert. Test cases with commit '7d331844' with Node_Type='sim' on Ubuntu Linux 16-04, and find following problems :

1.  9 test cases (8-1-01, 8-1-02, 8-2-01, 8-2-02, 9-2-04, 9-2-08, 9-2-12, 9-2-13, 9-2-14) failed with following error messages:
======================================================================
ERROR: test (__main__.Cert_8_1_01_Commissioning)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "Cert_8_1_01_Commissioning.py", line 62, in test
    self.nodes[COMMISSIONER].commissioner_start()
  File "/home/zjlu/work/openthread/tests/scripts/thread-cert/node.py", line 72, in commissioner_start
    self.interface.commissioner_start()
  File "/home/zjlu/work/openthread/tests/scripts/thread-cert/node_cli.py", line 145, in commissioner_start
    self.pexpect.expect('Done')
  File "/home/zjlu/.local/lib/python2.7/site-packages/pexpect/spawnbase.py", line 321, in expect
    timeout, searchwindowsize, async)
  File "/home/zjlu/.local/lib/python2.7/site-packages/pexpect/spawnbase.py", line 345, in expect_list
    return exp.expect_loop(timeout)
  File "/home/zjlu/.local/lib/python2.7/site-packages/pexpect/expect.py", line 107, in expect_loop
    return self.timeout(e)
  File "/home/zjlu/.local/lib/python2.7/site-packages/pexpect/expect.py", line 70, in timeout
    raise TIMEOUT(msg)
TIMEOUT: Timeout exceeded.
<pexpect.pty_spawn.spawn object at 0x7fd296570c10>
command: /home/zjlu/work/openthread/output/x86_64-unknown-linux-gnu/bin/ot-cli-ftd
args: ['/home/zjlu/work/openthread/output/x86_64-unknown-linux-gnu/bin/ot-cli-ftd', '1']
# buffer (last 100 chars): '\r\n> commissioner start\r\n> '
before (last 100 chars): '\r\n> commissioner start\r\n> '
after: <class 'pexpect.exceptions.TIMEOUT'>
match: None
match_index: None
exitstatus: None
flag_eof: False
pid: 496
child_fd: 5
closed: False
timeout: 4
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile("Done")

----------------------------------------------------------------------
Ran 1 test in 11.021s

FAILED (errors=1)
================================================

 These test cases are passed with the OT automation regression test (https://github.com/openthread/openthread/wiki/Thread-Certification), but it looks another set of test cases.  We are wondering if you passed all Cert. Test cases under '../tests/scripts/thread-cert' both on Linux and Windows?   

2. the 'sniffer.py' under ../tests/scripts/thread-cert/ uses Python 'select.poll' function which is not supported by Windows, we are wondering how to you run these Cert. Test cases under Windows? or which Library or version you are using for Windows?

Thanks,

zj

jwhui, 2016-12-02:
For issue (1), how did you configure your build?  Make sure you've enabled both the commissioner and joiner roles.  For example:

```
$ ./bootstrap
$ COMMISSIONER=1 JOINER=1 make -f examples/Makefile-cc2538
```

jwhui, 2016-12-02:
@prfi @nibanks, do either of you have experience running [test/scripts/thread-cert/sniffer.py](https://github.com/openthread/openthread/blob/master/tests/scripts/thread-cert/sniffer.py) on Windows?

nibanks, 2016-12-02:
No, I haven't ever run it on Windows.

bot-zjlu, 2016-12-02:
We ran the test cases under '../tests/scripts/thread-cert'  on Ubuntun
Linux with setting NODE_TYPE=sim.

Build with:

$ ./bootstrap
$ COMMISSIONER=1 JOINER=1 make -f examples/Makefile-posix

And we used to pass all the tests on Linux. but recently it no longer work.

jwhui, 2016-12-02:
FWIW, [Travis runs Ubuntu](https://docs.travis-ci.com/user/ci-environment/) and the most recent build checks [passed](https://travis-ci.org/openthread/openthread/builds/180822524).

How about trying:

```
$ ./bootstrap
$ make -f examples/Makefile-posix check
```

bot-zjlu, 2016-12-02:
it still failed, let me check with latest codes.
.
.
.
make  check-TESTS
FAIL: thread-cert/Cert_5_1_07_MaxChildCount.py
FAIL: thread-cert/Cert_5_2_01_BecomeActiveRouter.py
FAIL: thread-cert/Cert_5_2_04_REEDUpgrade.py
FAIL: thread-cert/Cert_5_2_02_LeaderReject1Hop.py
FAIL: thread-cert/Cert_5_2_03_LeaderReject2Hops.py
FAIL: thread-cert/Cert_5_2_07_REEDSynchronization.py
FAIL: thread-cert/Cert_5_2_06_RouterDowngrade.py
FAIL: thread-cert/Cert_5_2_05_AddressQuery.py
FAIL: thread-cert/Cert_5_3_01_LinkLocal.py
FAIL: thread-cert/Cert_5_1_04_RouterAddressReallocation.py
FAIL: thread-cert/Cert_5_1_08_RouterAttachConnectivity.py
FAIL: thread-cert/Cert_5_1_05_RouterAddressTimeout.py
FAIL: thread-cert/Cert_5_1_06_RemoveRouterId.py
FAIL: thread-cert/Cert_5_1_03_RouterAddressReallocation.py
FAIL: thread-cert/Cert_5_1_10_RouterAttachLinkQuality.py
FAIL: thread-cert/Cert_5_1_12_NewRouterNeighborSync.py
FAIL: thread-cert/Cert_5_1_11_REEDAttachLinkQuality.py
FAIL: thread-cert/Cert_5_3_02_RealmLocal.py
FAIL: thread-cert/Cert_5_3_03_AddressQuery.py
FAIL: thread-cert/Cert_5_3_06_RouterIdMask.py
FAIL: thread-cert/Cert_5_1_02_ChildAddressTimeout.py
FAIL: thread-cert/Cert_5_1_13_RouterReset.py
FAIL: thread-cert/Cert_5_1_01_RouterAttach.py
FAIL: thread-cert/Cert_5_5_02_LeaderReboot.py
FAIL: thread-cert/Cert_5_1_09_REEDAttachConnectivity.py
FAIL: thread-cert/Cert_5_3_07_DuplicateAddress.py
FAIL: thread-cert/Cert_5_3_05_RoutingLinkQuality.py
FAIL: thread-cert/Cert_5_3_08_ChildAddressSet.py
FAIL: thread-cert/Cert_5_5_04_SplitMergeRouters.py
FAIL: thread-cert/Cert_5_3_06b_RouterIdMask.py
FAIL: thread-cert/Cert_5_3_04_AddressMapCache.py
FAIL: thread-cert/Cert_5_5_01_LeaderReset.py
FAIL: thread-cert/Cert_5_5_07_SplitMergeThreeWay.py
FAIL: thread-cert/Cert_5_3_09_AddressQuery.py
FAIL: thread-cert/Cert_5_5_03_SplitMergeChildren.py
FAIL: thread-cert/Cert_5_3_10_AddressQuery.py
FAIL: thread-cert/Cert_5_6_01_NetworkDataRegisterBeforeAttachLeader.py
FAIL: thread-cert/Cert_5_5_08_SplitRoutersLostLeader.py
FAIL: thread-cert/Cert_5_5_06_SplitWeight.py
FAIL: thread-cert/Cert_5_6_07_NetworkDataRequestREED.py
FAIL: thread-cert/Cert_5_6_03_NetworkDataRegisterAfterAttachLeader.py
FAIL: thread-cert/Cert_5_5_05_SplitMergeREED.py
FAIL: thread-cert/Cert_5_6_09_NetworkDataForwarding.py
FAIL: thread-cert/Cert_5_6_08_ContextManagement.py
FAIL: thread-cert/Cert_5_6_02_NetworkDataRegisterBeforeAttachRouter.py
FAIL: thread-cert/Cert_5_6_04_NetworkDataRegisterAfterAttachRouter.py
FAIL: thread-cert/Cert_5_6_05_NetworkDataRegisterAfterAttachRouter.py
FAIL: thread-cert/Cert_5_8_03_KeyIncrementRollOver.py
FAIL: thread-cert/Cert_5_8_02_KeyIncrement.py
FAIL: thread-cert/Cert_5_6_06_NetworkDataExpiration.py
FAIL: thread-cert/Cert_6_1_03_RouterAttachConnectivity.py
FAIL: thread-cert/Cert_6_1_01_RouterAttach.py
FAIL: thread-cert/Cert_6_1_06_REEDAttachLinkQuality.py
FAIL: thread-cert/Cert_5_8_01_KeySynchronization.py
FAIL: thread-cert/Cert_6_1_02_REEDAttach.py
FAIL: thread-cert/Cert_6_2_01_NewPartition.py
FAIL: thread-cert/Cert_6_4_02_RealmLocal.py
FAIL: thread-cert/Cert_6_1_04_REEDAttachConnectivity.py
FAIL: thread-cert/Cert_6_1_05_RouterAttachLinkQuality.py
FAIL: thread-cert/Cert_6_4_01_LinkLocal.py
FAIL: thread-cert/Cert_7_1_02_BorderRouterAsRouter.py
FAIL: thread-cert/Cert_6_3_01_OrphanReattach.py
FAIL: thread-cert/Cert_6_2_02_NewPartition.py
FAIL: thread-cert/Cert_6_1_07_EDSynchronization.py
FAIL: thread-cert/Cert_6_6_01_KeyIncrement.py
FAIL: thread-cert/Cert_6_3_02_NetworkDataUpdate.py
FAIL: thread-cert/Cert_6_5_01_ChildResetSynchronize.py
FAIL: thread-cert/Cert_6_5_02_ChildResetReattach.py
FAIL: thread-cert/Cert_7_1_01_BorderRouterAsLeader.py
FAIL: thread-cert/Cert_8_1_02_Commissioning.py
FAIL: thread-cert/Cert_9_2_04_ActiveDataset.py
FAIL: thread-cert/Cert_7_1_03_BorderRouterAsLeader.py
FAIL: thread-cert/Cert_6_6_02_KeyIncrementRollOver.py
FAIL: thread-cert/Cert_8_1_01_Commissioning.py
FAIL: thread-cert/Cert_7_1_04_BorderRouterAsRouter.py
FAIL: thread-cert/Cert_7_1_05_BorderRouterAsRouter.py
FAIL: thread-cert/Cert_8_2_01_JoinerRouter.py
FAIL: thread-cert/Cert_9_2_08_PersistentDatasets.py
FAIL: thread-cert/Cert_8_2_02_JoinerRouter.py
FAIL: thread-cert/Cert_9_2_10_PendingPartition.py
FAIL: thread-cert/Cert_9_2_07_DelayTimer.py
FAIL: thread-cert/Cert_9_2_13_EnergyScan.py
FAIL: thread-cert/Cert_9_2_16_ActivePendingPartition.py
FAIL: thread-cert/Cert_9_2_12_Announce.py
FAIL: thread-cert/Cert_9_2_18_RollBackActiveTimestamp.py
FAIL: thread-cert/Cert_9_2_09_PendingPartition.py
FAIL: thread-cert/Cert_9_2_17_Orphan.py
FAIL: thread-cert/Cert_9_2_15_PendingPartition.py
FAIL: thread-cert/Cert_9_2_14_PanIdQuery.py
FAIL: thread-cert/Cert_9_2_11_MasterKey.py
make[7]: Nothing to be done for 'all'.
============================================================================
Testsuite summary for OPENTHREAD g7d33184
============================================================================
# TOTAL: 90
# PASS:  0
# SKIP:  0
# XFAIL: 0
# FAIL:  90
# XPASS: 0
# ERROR: 0
============================================================================
See tests/scripts/test-suite.log
Please report to openthread-devel@googlegroups.com
============================================================================
Makefile:918: recipe for target 'test-suite.log' failed
make[6]: *** [test-suite.log] Error 1
Makefile:1024: recipe for target 'check-TESTS' failed
make[5]: *** [check-TESTS] Error 2
Makefile:1720: recipe for target 'check-am' failed
make[4]: *** [check-am] Error 2
Makefile:436: recipe for target 'check-recursive' failed
make[3]: *** [check-recursive] Error 1
Makefile:591: recipe for target 'check-recursive' failed
make[2]: *** [check-recursive] Error 1
Makefile:884: recipe for target 'check' failed
make[1]: *** [check] Error 2
examples/Makefile-posix:257: recipe for target
'check-x86_64-unknown-linux-gnu' failed
make: *** [check-x86_64-unknown-linux-gnu] Error 2

bot-zjlu, 2016-12-02:
just verify with latest codes with commit: commit
917fd1d4cc77f9bf8d5f41e5d04f4962726266b2

Even I cant run './bootstrap', and it looks python version or some library
are different.

zj@ubuntu164-desktop:~/work/openthread-latest$ ./bootstrap
Unescaped left brace in regex is deprecated, passed through in regex;
marked by <-- HERE in m/\${ <-- HERE ([^ \t=:+{}]+)}/ at
/home/zjlu/work/openthread-latest/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/automake
line 3930.

zjlu@ubuntu164-desktop:~/work/openthread-latest$ python --version
Python 2.7.12

Thanks,

zj

jwhui, 2016-12-02:
The bootstrap issue is a compatibility issue with automake and newer versions of perl, as noted [here](http://serverfault.com/questions/731539/automake-error-unescaped-left-brace-in-regex-is-deprecated).  You could downgrade to an older version of perl (I'm using 5.18.2).  I'll look into how best to support newer versions of perl.

nibanks, 2016-12-06:
@jwhui What would be the exact command/process to downgrade to version 5.18.2?

bot-zjlu, 2016-12-06:
I use perlbrew <https://perlbrew.pl/> to downgrade Perl from 5.22.1 to
5.18.2.  But still failed on bootstrap.  But on Unbuntu-13-04 with Python
2.7.5 and Perl-5.14.2, bootstrap works fine.

========================== screen output
======================================
zj@ubuntu164-desktop:~/work$ cd openthread-latest
zj@ubuntu164-desktop:~/work/openthread-latest$ ./bootstrap
Unescaped left brace in regex is deprecated, passed through in regex;
marked by <-- HERE in m/\${ <-- HERE ([^ \t=:+{}]+)}/ at
/home/zj/work/openthread-latest/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/automake
line 3930.
zj@ubuntu164-desktop:~/work/openthread-latest$ perl --version

This is perl 5, version 18, subversion 2 (v5.18.2) built for x86_64-linux
(with 1 registered patch, see perl -V for more detail)

Copyright 1987-2013, Larry Wall

Perl may be copied only under the terms of either the Artistic License or
the
GNU General Public License, which may be found in the Perl 5 source kit.

Complete documentation for Perl, including FAQ lists, should be found on
this system using "man perl" or "perldoc perl".  If you have access to the
Internet, point your browser at http://www.perl.org/, the Perl Home Page.

===========================================================

bot-zjlu, 2016-12-07:
We tried the './bootstrap' in latest OT sources with Ubuntu_13.10 with
default Python 2.7.9 and Perl 5.14.2, it works.

http://stackoverflow.com/questions/31753737/deprecated-left-curly-bracket-in-perl-regex-exactly-when

Also when set 'COVERAGE != 1' instead of 'COVERAGE ?= 0', the 9 failed
tests with Commissioner response error are passed.

But we still can't run Cert. Tests on Windows platform because Python
doesn't support 'select.poll'. But we see in the source codes, OT
automation test should run under Windows and there is a window native test
runner, how do you run the Cert. Test cases on Windows?

Thanks,

zj

nibanks, 2016-12-08:
That native test running is only for testing on virtual Windows networking devices. I have not been able to get all the code publicly available for that yet. I hope to in the coming weeks.

jwhui, 2016-12-09:
@bot-zjlu, submitted a proposed fix in #1060.

closed 2016-12-14 by jwhui