openthread/openthread #1040: 6LBR like smart bridge mode

mythay, 2016-12-05:
I have learned that the IP tunnel is the recommended  communication mode between the host CPU and Openthread device (such as wpantund). It is fine for the CPU which can run complex OS like Linux and Windows, but for resource constraint MCU, it is hard to use.

I know that [6LBR](https://github.com/cetic/6lbr) can support not only tunnel mode, but also other mode. If I use the smart bridge mode, I can use cc2538 to connect to an enc28j60 chip and let this Openthread device as a simple Ethernet device.

Can Openthread supports such mode?

jwhui, 2016-12-05:
I'm not sure what you mean by "smart bridge" mode, that is not a traditional networking term.  [Bridging](https://en.wikipedia.org/wiki/Bridging_(networking)) is not generally recommended given the resources disparities between Thread and other link technologies like WiFi and Ethernet.  Instead, [routers](https://en.wikipedia.org/wiki/Router_(computing)) should be used to connect Thread networks to other IPv6 networks.

In any case, OpenThread provides an API to send/receive raw IPv6 datagrams, which may be used for routing IPv6 datagrams to/from other IPv6 networks.  The [NCP](https://github.com/openthread/openthread/tree/master/src/ncp) implementation uses this capability to transfer IPv6 datagrams to/from [wpantund](https://github.com/openthread/wpantund/).

```
ThreadError otSendIp6Datagram(otInstance *aInstance, otMessage aMessage);

void otSetReceiveIp6DatagramCallback(otInstance *aInstance, otReceiveIp6DatagramCallback aCallback,
                                     void *aCallbackContext);
```

mythay, 2016-12-06:
@jwhui I understand your idea that router is a better way, and I think it is why another famous 6lowpan project Contiki only support slip officially.

But 6LBR provide more choice for border router, the router can be very simple and don't need a powerful host CPU running Linux or windows.

I extract some words and pictures from 6LBR as a brief introduction.

> The SmartBridge is acting as a NDP proxy on the Ethernet side and is using NDP parameters to configure the WSN mesh. Source and destination MAC addresses are translated and addresses present in ICMPv6 packets are also translated. 
> [image]

[The original link](https://github.com/cetic/6lbr/wiki/6LBR-Modes#the-seven-modes-in-more-detail)

jwhui, 2016-12-06:
Thanks for the pointers.  From what I understand, "SmartBridge" is essentially a marketing term for using IPv6 ND Proxy to bridge IEEE 802.15.4 networks.  Such methods work with RPL non-storing networks because the RPL root maintains state of all devices in its subtree.

In contrast, Thread does not require a single node to maintain state about all devices in the network.  This is by design, to increase robustness (i.e. no single-point-of-failure) and reduce resource requirements on any one device in the network.  As a result, Thread by itself does not directly support an IPv6 ND Proxy model.  However, it is possible to layer on additional application-layer functionality (e.g. DHCPv6, unsolicited Address Notifications, application-specific messaging, etc.) that can be used to provide such information to a bridging device.  OpenThread does not currently implement such functionality, but it can be done.

mythay, 2016-12-06:
Is there any plan for such features?

jwhui, 2016-12-08:
@mythay, we do not currently have any formal plan for such features.  However, we are always open to contributions.

closed 2016-12-14 by jwhui