openthread/openthread #1067: sniffer.py Assumes Posix Socket Virtual Environment

nibanks, 2016-12-13:
Since `sniffer.py` and the associated functions were added to the certification tests, all tests have been broken on Windows. Extensive effort was put into refactoring `node.py` to allow for the tests to run on both Posix socket version of `otCli` and the `otNodeApi.dll` exposed on Windows.

A solution needs to be found to fix these issues. At the very least we should have the ability to disable the `sniffer.py` parts of the certification tests.

A better design would perhaps make a new interface in the `node` class for a node in 'sniffer' mode. It could then expose some functions for allowing the sniffer to get the byte buffers from that? For the Windows version, I could then implement the corresponding versions.

jwhui, 2016-12-13:
I agree it would be nice for the test suite to support different platforms.  A similar issue was brought up as part of #1036.

@prfi, thoughts?

prfi, 2016-12-14:
I totally agree that tests should support different platforms. I've made changes that should allow to meet this requirement. Please check my pull request and let me know  if provided modifications are ok or not. If you have any idea how to improve sniffer design also let me know.

jwhui, 2016-12-14:
Fixed by #1073.

closed 2016-12-14 by jwhui