openthread/openthread #1068: test_ipv6.py: spurious test errors

jwhui, 2016-12-13:
We're seeing some spurious test failures on Travis related to `test_ipv6.py`.  Example build log below.

@prfi, can you take a look?

```bash
...

make  check-TESTS
PASS: thread-cert/test_common.py
FAIL: thread-cert/test_ipv6.py
PASS: thread-cert/test_network_layer.py
PASS: thread-cert/test_crypto.py
PASS: thread-cert/test_coap.py
PASS: thread-cert/test_network_data.py

...

make[8]: Nothing to be done for `all'.
=======================================================
   OPENTHREAD g0b1d47a: tests/scripts/test-suite.log
=======================================================

# TOTAL: 99
# PASS:  98
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

FAIL: thread-cert/test_ipv6.py
==============================

.E................................................
======================================================================
ERROR: test_should_create_HopByHop_object_instance_when_to_bytes_method_is_called_with_data (__main__.TestHopByHopFactory)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "../../../tests/scripts/thread-cert/test_ipv6.py", line 955, in test_should_create_HopByHop_object_instance_when_to_bytes_method_is_called_with_data
    self.assertEqual(hop_by_hop_option.value.S, hop_by_hop.options[0].value.S)
IndexError: list index out of range

----------------------------------------------------------------------
Ran 50 tests in 0.037s

FAILED (errors=1)
FAIL thread-cert/test_ipv6.py (exit status: 1)

============================================================================
Testsuite summary for OPENTHREAD g0b1d47a
============================================================================
# TOTAL: 99
# PASS:  98
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
============================================================================
See tests/scripts/test-suite.log
Please report to openthread-devel@googlegroups.com
============================================================================
make[7]: *** [test-suite.log] Error 1
make[6]: *** [check-TESTS] Error 2
make[5]: *** [check-am] Error 2
make[4]: *** [check-recursive] Error 1
make[3]: *** [check-recursive] Error 1
make[2]: *** [check] Error 2
make[1]: *** [distcheck] Error 1
make: *** [distcheck-x86_64-unknown-linux-gnu] Error 2
+die
+echo  *** ERROR: 
 *** ERROR: 
+exit 1

The command ".travis/script.sh" exited with 1.

Done. Your build exited with 1.
```

prfi, 2016-12-14:
I will try to fix this today

jwhui, 2016-12-14:
Fixed by #1072.

closed 2016-12-14 by jwhui