openthread/openthread #1093: Thread Harness - 8.1.1(OMC Joining,no JR,any commissioner,single(correct)) failure issue with ARM mbed Golden device

charishmag, 2016-12-20:
Hi,
   Test 8.1.1(OMC Joining,no JR,any commissioner,single(correct)) is failing with ARM: NXP FRDM-K64F with Firefly as a Golden Device, whereas similar test is passing with openthread device taken as Golden device.
   ARM: NXP FRDM-K64F as a golden device is not responding for **Discovery request** from openthread device as a Router. 
   Please find the attached image with no response for Discovery Request.
[image]

jwhui, 2016-12-20:
@charishmag, thanks for pointing this out.  We'll take a look.  Thanks.

librasungirl, 2017-02-09:
Hi @charishmag, this issue should have been fixed recently. Will close it now.
Please retry with latest master branch code and feel free to reopen it if you still have the problem. Thanks.

closed 2017-02-09 by librasungirl