openthread/openthread #1100: OpenThread API Naming Cleanup

jwhui, 2016-12-22:
I'd like to clean up the OpenThread API naming.

Some issue I see with what exists today:
- Not grouped by component (although, some examples of grouping by component does exist, such as `otCoap*`).
- Some boolean properties are get/set while others are enable/disable.
- Implementation is in one large file `src/openthread.cpp`.
- API header files use `include/openthread-` prefix, while platform API header filenames are `include/platform/` prefix.

Proposed convention:

- API Naming Convention
  - Commands
    - ot[Component][Command] (e.g. `otMacEnergyScan(...)`)
  - Properties
    - ot[Component]Get[Property] (e.g. `otMacGetChannel(...)`)
    - ot[Component]Set[Property] (e.g. `otMacSetChannel(...)`)

- File Naming Convention
  - Client API header files
    - include/ot-api-[component].h (e.g. `include/ot-api-mac.h`)
  - Platform API header files
    - include/ot-plat-[component].h (e.g. `include/ot-plat-radio.h`)
  - API implementation files
    - src/api/[component].cpp: (e.g. `src/api/mac.cpp`)

Thoughts and feedback appreciated.

nibanks, 2016-12-22:
That all sounds pretty good to me. Just to confirm, that means you won't be having any more subfolders in `include`?

For your example of `include/ot-api-mac.h` is that supposed to be what currently is `src/core/mac/mac.hpp`?

LuDuda, 2016-12-22:
LGTM :+1:

For boolean properties i would prefer to have one Set function instead of two Enable and Disable.

Also I see that OTAPI and OTCALL keys which were added to support Windows are present only in openthread.h and not with other file like openthread-ip6.h.

@nibanks I believe in `include/ot-api-mac.h `we should have declarations related to MAC functionality from current `openthread.h` and not to expose our internal `core/mac/mac.hpp` calls.

nibanks, 2016-12-22:
@LuDuda I agree that `include/ot-api-mac.h` should be that. I just wanted to confirm. I also prefer one set instead of enable/disable.

As for the `OTAPI` & `OTCALL`, the reason I only added it to some is because those are the control interfaces (not datapath related) and are the only ones exposed to user mode. For the sake of completeness though, it doesn't matter to me if we put them everywhere.

turon, 2016-12-22:
I'm a very adamant supporter of the API naming portion of this proposal!

Is the motivation for flattening the include directories to eliminate namespace collisions with things like #include "platform/radio.h"?

jwhui, 2016-12-22:
@nibanks, header files in `include` should only expose public APIs (i.e. those in `openthread.h`).

jwhui, 2016-12-22:
An alternative to the header file naming convention is to add `openthread` to the path instead of the filename.

Client API header files
- include/openthread/[component].h (e.g. `include/openthread/mac.h`)

Platform API header files
- include/openthread/platform/[component].h (e.g. `include/openthread/platform/radio.h`)

turon, 2016-12-22:
I prefer using a top-level directory such as "openthread" or "ot" for header namespace management to the flat, hyphenated ot-module-foo.h model. 

```
#include <openthread/types.h>
#include <openthread/mac.h>
#include <openthread/platform/radio.h>
```

nibanks, 2016-12-22:
I prefer the hyphenated approach so that the file names are generally unique when compared to other libraries.

jwhui, 2017-03-10:
Resolved in #1173.

closed 2017-03-10 by jwhui