openthread/openthread #1137: Python Tests: No module named 'Crypto'

nibanks, 2017-01-12:
I am working on getting the Python tests running on Windows again, since the dependency on the sniffer interface was added. I am coming across a dependency I am lacking but I don't know where/how to install it. Any help would be appreciated.

```
  File "scripts\Cert_5_1_01_RouterAttach.py", line 33, in <module>
    import config
  File "C:\thread\scripts\config.py", line 33, in <module>
    import message
  File "C:\thread\scripts\message.py", line 39, in <module>
    import mac802154
  File "C:\thread\scripts\mac802154.py", line 39, in <module>
    from net_crypto import AuxiliarySecurityHeader, CryptoEngine, MacCryptoMaterialCreator
  File "C:\thread\scripts\net_crypto.py", line 36, in <module>
    from Crypto.Cipher import AES
ModuleNotFoundError: No module named 'Crypto'
```

jwhui, 2017-01-12:
The package you need is pycryptodome version 3.4.3.   You can install it via the following command.

```bash
pip install pycryptodome==3.4.3
```

I ran into this issue as part of #985.  I agree this should be included in a README or doc somewhere.

For future reference, viewing the Travis [before_install.sh](https://github.com/openthread/openthread/blob/master/.travis/before_install.sh) script is also helpful.

nibanks, 2017-01-12:
Thanks. That seemed to work!

closed 2017-01-12 by nibanks

zsolthajnal, 2018-01-26:
Thanks as well. This worked to me too.

Macielyoung, 2018-04-09:
Thanks. I have tried various methods and they didn't work to me. Your method works well.

jwhui, 2018-07-03:
@guanhui07, uninstall pycrypto.  There is name collision in packages `pycrypto` and `pycryptodome` (both have module `Crypto`).

cisordeng, 2018-08-08:
Maybe to python scripts list rename `Crypto` to `crypto` or use `crypto`.
haha,My English is so poor.

cisordeng, 2018-08-08:
haha,Thanks.

PandaTea, 2018-11-03:
amazing,its work

sreeramp96, 2019-05-09:
from Crypto.PublicKey import RSA
ModuleNotFoundError: No module named 'Crypto'
I have installed both crypto and Crypto using pip, but still it shows this error

HarshdipD, 2019-09-14:
I'm getting the same error, and I have pycrpytodome installed version 3.9.0

jwhui, 2019-09-17:
@HarshdipD, did you uninstall `pycrypto`?

If so, can you provide the output of `pip list | grep crypt`?

HarshdipD, 2019-09-17:
@jwhui  I did not have pycrypto installed. My issue was solved by changing folder name from `crypto` to `Crypto`. Thank you!

blackfur, 2019-10-12:
> Public

Have you solve this? I have this error too.

saikiran994, 2019-10-27:
even i am also facing same problem

bukepo, 2019-10-27:
@blackfur @saikiran994 Could you try `pip3 install -r tests/scripts/thread-cert/requirements.txt`?

saikiran994, 2019-10-28:
requriments already satisfied but not import lib to work, still showing no
module found

jwhui, 2019-10-28:
@saikiran994, have you tried the proposed resolutions above?

saikiran994, 2019-10-29:
yeahhhh.....so i tried in windows in that also getting same

jwhui, 2019-10-29:
@saikiran994, to verify:

1. `pycrypto` is **not** installed
2. `pycryptodome` is installed

If that does not help, can you provide the output of `pip3 list | grep crypt`?

jdougal84, 2019-10-31:
Had the same issue on Win10. 

Rename the folder in site-packages from crypto to Crypto.

4tal, 2019-11-19:
> @saikiran994 Had the same issue on Win10.
> 
> Rename the folder in site-packages from crypto to Crypto.

Help me as well on Mac

68jai90, 2020-04-02:
> @jwhui I did not have pycrypto installed. My issue was solved by changing folder name from `crypto` to `Crypto`. Thank you!

This worked for me to

simonlingoogle, 2020-04-03:
> > @jwhui I did not have pycrypto installed. My issue was solved by changing folder name from `crypto` to `Crypto`. Thank you!
> 
> This worked for me to

If this is the solution, maybe we should try import both Crypto and crypto and choose whichever works?

2s5a3i8d, 2020-04-17:
I have the same problem. I have installed python via anaconda. Where can I change folder name from crypto to Crypto?

AnrixarApps, 2020-05-17:
Ok i found solution for this
first of all use command >>pip uninstall crypto
then >> pip uninstall pycryptodome
then >> pip install pycryptodome

Mariiooo, 2020-05-26:
Yo, I have this problem now with in our Python application.
We are using Crypto. On my friends MAC it's working. But when I open the code on WINDOWS I'm getting an error.

from Crypto.Cipher import AES
ModuleNotFoundError: No module named 'Crypto'

This is my Pip List.
Package      Version
------------ ----------
certifi      2020.4.5.1
chardet      3.0.4
idna         2.9
Naked        0.1.31
pip          20.1.1
pycryptodome 3.9.7
PyYAML       5.3.1
requests     2.23.0
setuptools   41.2.0
shellescape  3.8.1
urllib3      1.25.9
wheel        0.34.2

What's the problem and how can I solve it?

68jai90, 2020-05-26:
Try to change "Crypto" to "crypto". This is how is solved my issue.

Mariiooo, 2020-05-26:
> Try to change "Crypto" to "crypto". This is how is solved my issue.
> […](#)

Not working as well mate :( 

If i change it to crypto it's still giving error's
[image]

sameersam97, 2020-05-29:
You just need to uninstall all your previous versions of crypto and pycryptodome and have a fresh installation of pycryptodome alone. 

This works for me, Hope so it works for you as well :)

pip uninstall crypto
pip uninstall pycryptodome
pip install pycryptodome

Use this link for reference
-----------------------------------------
https://pycryptodome.readthedocs.io/en/latest/src/faq.html#why-do-i-get-the-error-no-module-named-crypto-on-windows

> > Try to change "Crypto" to "crypto". This is how is solved my issue.
> > […](#)
> 
> Not working as well mate :(
> 
> If i change it to crypto it's still giving error's
> [image]
> 
> Here is what i now have with the from Crypto.Cipher import AES
> 
> [image]

You just need to uninstall all your previous versions of crypto and pycryptodome and have a fresh installation of pycryptodome alone. 

This works for me, Hope so it works for you as well :)

pip uninstall crypto
pip uninstall pycryptodome
pip install pycryptodome

Use this link for reference
-----------------------------------------
https://pycryptodome.readthedocs.io/en/latest/src/faq.html#why-do-i-get-the-error-no-module-named-crypto-on-windows

Mariiooo, 2020-05-29:
It has been fixed :)  Our pipeline docker container was not correctly set up.  So that's where the errors came from.  

It had nothing to do with Crypto :P

Basha963, 2020-06-02:
> You just need to uninstall all your previous versions of crypto and pycryptodome and have a fresh installation of pycryptodome alone.
> 
> This works for me, Hope so it works for you as well :)
> 
> pip uninstall crypto
> pip uninstall pycryptodome
> pip install pycryptodome
> 
> ## Use this link for reference
> https://pycryptodome.readthedocs.io/en/latest/src/faq.html#why-do-i-get-the-error-no-module-named-crypto-on-windows
> 
> > > Try to change "Crypto" to "crypto". This is how is solved my issue.
> > > […](#)
> > 
> > 
> > Not working as well mate :(
> > If i change it to crypto it's still giving error's
> > [image]
> > Here is what i now have with the from Crypto.Cipher import AES
> > [image]
> 
> You just need to uninstall all your previous versions of crypto and pycryptodome and have a fresh installation of pycryptodome alone.
> 
> This works for me, Hope so it works for you as well :)
> 
> pip uninstall crypto
> pip uninstall pycryptodome
> pip install pycryptodome
> 
> ## Use this link for reference
> https://pycryptodome.readthedocs.io/en/latest/src/faq.html#why-do-i-get-the-error-no-module-named-crypto-on-windows

Thank you this works for me

BhavsinhBapu, 2020-06-15:
I have Solve This Problem to This Way

first of all use command >>pip uninstall crypto
then >> pip uninstall pycryptodome
then >> pip install pycryptodome

HMaliniV, 2020-07-26:
I have the same error - ModuleNotFoundError: No module named 'Crypto'
i uninstalled pycryptodome and did 
pip install pycryptodome==3.4.3

I am unable to install pycrypto 
please help !!

jamescalam, 2020-07-30:
> I have the same error - ModuleNotFoundError: No module named 'Crypto'
> i uninstalled pycryptodome and did
> pip install pycryptodome==3.4.3
> 
> I am unable to install pycrypto
> please help !!

@HMaliniV try the most recent pycryptodome, eg do this:
```
pip uninstall crypto
pip uninstall pycryptodome
pip install pycryptodome
```

santhoshpsk4, 2020-08-10:
> Ok i found solution for this
> first of all use command >>pip uninstall crypto
> then >> pip uninstall pycryptodome
> then >> pip install pycryptodome

The best known solution. Worked actually.

loginov671, 2020-08-19:
> @HMaliniV try the most recent pycryptodome, eg do this:
>
> ```
> pip uninstall crypto
> pip uninstall pycryptodome
> pip install pycryptodome
> ```

don`t work for me...

imax16041998, 2021-01-18:
i have same error. **`ModuleNotFoundError: No module named 'Crypto'`**
I installed **cryptodome** v3.9.9 and uninstall package **crypto** 
How to fix it????
[image]

my email: imax16041998@gmail.com
thank you so much!

jwhui, 2021-01-18:
@imax16041998 , did you try the proposed solutions that were posted above in this issue?

imax16041998, 2021-01-18:
@jwhui i tried but still error.
i tried uninstall Crypto and install cryptodome

jwhui, 2021-01-18:
@imax16041998 , did you try uninstalling both crypto and pycryptodome first and then reinstalling pycryptodome?
```
pip uninstall crypto
pip uninstall pycryptodome
pip install pycryptodome
```

imax16041998, 2021-01-18:
@jwhui yes. i tried but still error :(

imax16041998, 2021-01-18:
can you give me your email and help me run my souce code?

Irving-cl, 2021-01-19:
@imax16041998 

The python environment is sometimes troublesome. If you want to run python tests, I would recommend running it using docker. Would you consider that? I can provide some help.

Irving-cl, 2021-01-19:
@imax16041998 

Just noticed that you work on windows system. I haven't tried the docker solution on windows. I think it can also work.

zisis912, 2021-02-26:
thank you so much!

iniakpothompson, 2021-05-02:
These were my steps to solve the problem:
1. pip install crypto
2. pip install pycryptodome
3. Change crypto to Crypto in the Library->site-packages folder of my virtual environment

selva49, 2021-05-18:
Traceback (most recent call last):
  File "/home/kali/Desktop/Veil/./Veil.py", line 98, in <module>
    the_conductor = orchestra.Conductor(args)
  File "/home/kali/Desktop/Veil/lib/common/orchestra.py", line 29, in __init__
    self.load_tools(cli_stuff)
  File "/home/kali/Desktop/Veil/lib/common/orchestra.py", line 75, in load_tools
    self.imported_tools[name] = module.Tools(
  File "tools/evasion/tool.py", line 39, in __init__
    self.load_payloads(cli_options)
  File "tools/evasion/tool.py", line 248, in load_payloads
    module = helpers.load_module(name)
  File "/home/kali/Desktop/Veil/lib/common/helpers.py", line 172, in load_module
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "tools/evasion/payloads/auxiliary/pyinstaller_wrapper.py", line 9, in <module>
    from tools.evasion.evasion_common import encryption
  File "/home/kali/Desktop/Veil/tools/evasion/evasion_common/encryption.py", line 8, in <module>
    from Crypto.Cipher import ARC4
ModuleNotFoundError: No module named 'Crypto'

Please help we

selva49, 2021-05-18:
ThankYou I got solution

tapaisystem, 2021-05-19:
pip uninstall crypto 
pip uninstall pycrypto 
pip install pycryptodome
work on me

jwsheen, 2021-06-24:
I resolved same that.
Just I rename "C:\Users\{User}\{Project Folder}\venv\lib\site-packages\crypto" to  "C:\Users\{User}\{Project Folder}\venv\lib\site-packages\Crypto".

Why ?
Crypto\Cypher\__init__.py use like:

from Crypto.Cipher._mode_ecb import _create_ecb_cipher
from Crypto.Cipher._mode_cbc import _create_cbc_cipher
from Crypto.Cipher._mode_cfb import _create_cfb_cipher
from Crypto.Cipher._mode_ofb import _create_ofb_cipher
from Crypto.Cipher._mode_ctr import _create_ctr_cipher
from Crypto.Cipher._mode_openpgp import _create_openpgp_cipher
from Crypto.Cipher._mode_ccm import _create_ccm_cipher
from Crypto.Cipher._mode_eax import _create_eax_cipher
from Crypto.Cipher._mode_siv import _create_siv_cipher
from Crypto.Cipher._mode_gcm import _create_gcm_cipher
from Crypto.Cipher._mode_ocb import _create_ocb_cipher

ArpitArtworks, 2021-06-24:
> Ok i found solution for this
> first of all use command >>pip uninstall crypto
> then >> pip uninstall pycryptodome
> then >> pip install pycryptodome

It simply works!!

rdxpt, 2021-09-01:
Has anyone ran into cases where installing pycryptodome results only in the dist.info being installed, not the actual library source? If so, how did you resolve it? Tried a few different approaches above without any results to solve this specific issue.

Facing this right now on Ubuntu 18.04 on AWS EC2; happy to give more info if needed.

saleems11, 2021-09-02:
> These were my steps to solve the problem:
> 
> 1. pip install crypto
> 2. pip install pycryptodome
> 3. Change crypto to Crypto in the Library->site-packages folder of my virtual environment

Solved my problem Thanks alot.

SsNiPeR1, 2021-11-26:
> The package you need is pycryptodome version 3.4.3. You can install it via the following command.
> 
> ```shell
> pip install pycryptodome==3.4.3
> ```
> 
> I ran into this issue as part of #985. I agree this should be included in a README or doc somewhere.
> 
> For future reference, viewing the Travis [before_install.sh](https://github.com/openthread/openthread/blob/master/.travis/before_install.sh) script is also helpful.

didn't work for me.

zisis912, 2021-12-11:
Oh yeah I forgot to say, the solutions worked

gonchik, 2021-12-16:
Magic it does not work for me on old mac, but it works on M1.

pepa65, 2023-01-30:
This fixed it for me:
`ln -s /usr/lib/python3/dist-packages/Cryptodome /usr/lib/python3/dist-packages/Crypto`

scavro, 2024-04-30:
> This fixed it for me: `ln -s /usr/lib/python3/dist-packages/Cryptodome /usr/lib/python3/dist-packages/Crypto`

Thanks

scavro, 2024-04-30:
Thanks [pepa6](https://github.com/pepa65)

scavro, 2024-04-30:
Thanks [pepa65](https://github.com/pepa65)