openthread/openthread #1146: Python Certification Tests Don't Handle Extra Messages

nibanks, 2017-01-13:
When running the Python certification tests on Windows, the interface is open to every application/service that runs on Windows. So, the second it become available many different things try to use it. This is breaking the Python certification tests that assume a 'perfect' scenario of just the messages from the OpenThread stack.

Here is an example of an exception I am getting because the Windows stack sends a packet the Python code doesn't understand:

```
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "C:\Users\admin\AppData\Local\Programs\Python\Python36\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "C:\Users\admin\AppData\Local\Programs\Python\Python36\lib\threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "C:\thread\scripts\sniffer.py", line 85, in _sniffer_main_loop
    msg = self._message_factory.create(io.BytesIO(data))
  File "C:\thread\scripts\message.py", line 376, in create
    ipv6_packet = self._lowpan_parser.parse(lowpan_payload, message_info)
  File "C:\thread\scripts\lowpan.py", line 1093, in parse
    return self._handle_iphc_header(data, message_info)
  File "C:\thread\scripts\lowpan.py", line 1073, in _handle_iphc_header
    return self._ipv6_packet_factory.parse(io.BytesIO(decompressed_data), message_info)
  File "C:\thread\scripts\ipv6.py", line 876, in parse
    upper_layer_protocol = self._parse_upper_layer_protocol(data, next_header, message_info)
  File "C:\thread\scripts\ipv6.py", line 866, in _parse_upper_layer_protocol
    return factory.parse(data, message_info)
  File "C:\thread\scripts\ipv6.py", line 1037, in parse
    factory = self._get_icmpv6_body_factory(header.type)
  File "C:\thread\scripts\ipv6.py", line 1032, in _get_icmpv6_body_factory
    raise RuntimeError("Could not find factory to parse ICMP body. Unsupported ICMP type: {}".format(_type))
RuntimeError: Could not find factory to parse ICMP body. Unsupported ICMP type: 133
```

This is preventing all Python certification tests from being run on Windows.

prfi, 2017-01-16:
Maybe I'm wrong but I think that assumption is that only OpenThread stacks and sniffer takes a part in the tests. I think that this kind of tests should give an answer if there is no regression in stack behavior and they shouldn't be disturbed by the external services. To be consistent with this assumption I break test always when the unexpected messages appear. Anyhow I have made changes that should allow you to run this tests in your environment.

nibanks, 2017-01-16:
One of the reasons to test it on Windows is that Windows provides additional stress on the system. Besides that, there is no way to test it on Windows, as an actual network interface, without the entire system being able to bind to it.

nibanks, 2017-01-30:
I just hit another type of failure:

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\thread\scripts\ipv6.py", line 893, in _get_HopByHopOption_value_factory
    return self._options_factories[_type]
KeyError: 5
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "C:\Python\Python36\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "C:\Python\Python36\lib\threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "C:\thread\scripts\sniffer.py", line 83, in _sniffer_main_loop
    msg = self._message_factory.create(io.BytesIO(data))
  File "C:\thread\scripts\message.py", line 376, in create
    ipv6_packet = self._lowpan_parser.parse(lowpan_payload, message_info)
  File "C:\thread\scripts\lowpan.py", line 1087, in parse
    return self._handle_first_fragmentation_header(data, message_info)
  File "C:\thread\scripts\lowpan.py", line 1003, in _handle_first_fragmentation_header
    ipv6_header, extension_headers, udp_header = self._decompress_iphc(data, message_info)
  File "C:\thread\scripts\lowpan.py", line 995, in _decompress_iphc
    return self._lowpan_decompressor.decompress(data, message_info)
  File "C:\thread\scripts\lowpan.py", line 793, in decompress
    extension_header = self._lowpan_extension_headers_factory.parse(data, message_info)
  File "C:\thread\scripts\lowpan.py", line 257, in parse
    return factory.parse(data, next_header, message_info)
  File "C:\thread\scripts\lowpan.py", line 217, in parse
    options = self._hop_by_hop_options_factory.parse(io.BytesIO(ext_header_data), message_info)
  File "C:\thread\scripts\ipv6.py", line 912, in parse
    factory = self._get_HopByHopOption_value_factory(option_header.type)
  File "C:\thread\scripts\ipv6.py", line 895, in _get_HopByHopOption_value_factory
    raise RuntimeError("Could not find HopByHopOption value factory for type={}.".format(_type))
RuntimeError: Could not find HopByHopOption value factory for type=5.
```

@prfi, could you take a look please?

jwhui, 2017-03-10:
@nibanks, is this still an issue for you?

nibanks, 2017-03-10:
I still get those errors, but I silently ignore them. I'd prefer the code that handles the parsing didn't throw the exception, but I'm not blocked by it.

jwhui, 2017-03-10:
@prfi, thoughts?

prfi, 2017-03-15:
In my opinion test framework should raise exception when it get an unsupported message (it has implemented support for all messages required by the Thread). But I am fine with leave it as is and close this issue.

jwhui, 2017-07-13:
Closing stale issue.

closed 2017-07-13 by jwhui